isaac-sim/IsaacLab #2255: ModuleNotFoundError: No module named 'omni.kit.usd'

agj-neu, 2025-04-05:
### Question

Basic questions, related t(gru_sim) ubuntu2004@ubuntu2004-1234:~/isaac-sim4.5/IsaacLab$ python ./scripts/tutorials/00_sim/create_empty.py
[WARN][AppLauncher]: There are no arguments attached to the ArgumentParser object. If you have your own arguments, please load your own arguments before calling the `AppLauncher.add_app_launcher_args` method. This allows the method to check the validity of the arguments and perform checks for argument names.
[INFO][AppLauncher]: Loading experience file: /home/ubuntu2004/isaac-sim4.5/IsaacLab/apps/isaaclab.python.kit
Loading user config located at: '/home/ubuntu2004/issac_sim/kit/data/Kit/Isaac-Sim/4.5/user.config.json'
[Info] [carb] Logging to file: /home/ubuntu2004/issac_sim/kit/logs/Kit/Isaac-Sim/4.5/kit_20250405_234332.log
2025-04-05 15:43:32 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.096s] Failed to solve some dependencies locally, syncing with extension registry...
[0.097s] [ext: omni.kit.async_engine-0.0.1] startup
[0.100s] [ext: omni.client-1.2.0] startup
2025-04-05T15:43:32.500178Z  INFO main ThreadId(01) hub: hub/src/hub.rs:54: return=Ok(())
[0.175s] [ext: omni.kit.registry.nucleus-0.0.0] startup
2025-04-05 15:43:37 [5,397ms] [Error] [omni.ext.plugin] Failed to resolve extension dependencies. Failure hints:
	[isaaclab.python-2.0.2] dependency: 'isaacsim.asset.importer.mjcf' = { version='^' } can't be satisfied. Available versions:
	 (none found)
 Platform incompatible packages:
	 - [isaacsim.asset.importer.mjcf-2.3.7+106.5.0.lx64.d.cp310] (registry)
	 - [isaacsim.asset.importer.mjcf-2.3.7+106.5.0.wx64.d.cp310] (registry)
	 - [isaacsim.asset.importer.mjcf-2.3.7+106.5.0.lx64.r.cp310] (registry)
	 - [isaacsim.asset.importer.mjcf-2.3.7+106.5.0.wx64.r.cp310] (registry)
	 - [isaacsim.asset.importer.mjcf-2.3.4+106.5.0.lx64.r.cp310] (registry)
	 - [isaacsim.asset.importer.mjcf-2.3.4+106.5.0.wx64.r.cp310] (registry)
	 - [isaacsim.asset.importer.mjcf-2.3.4+106.5.0.wx64.d.cp310] (registry)
	 - [isaacsim.asset.importer.mjcf-2.3.4+106.5.0.lx64.d.cp310] (registry)
	 - [isaacsim.asset.importer.mjcf-2.3.4+107.0.3.lx64.r.cp311] (registry)
	 - [isaacsim.asset.importer.mjcf-2.3.4+107.0.3.wx64.d.cp311] (registry)
	 - ... (2 more) ...
 Synced registries:
	 - kit/default         : found 831 packages 
	 - kit/sdk             : found 324 packages 
	 - kit/community       : found 409 packages 

2025-04-05 15:43:37 [5,397ms] [Error] [omni.kit.app.plugin] Exiting app because of dependency solver failure...
Traceback (most recent call last):
  File "/home/ubuntu2004/isaac-sim4.5/IsaacLab/./scripts/tutorials/00_sim/create_empty.py", line 29, in <module>
    app_launcher = AppLauncher(args_cli)
  File "/home/ubuntu2004/isaac-sim4.5/IsaacLab/source/isaaclab/isaaclab/app/app_launcher.py", line 114, in __init__
    self._create_app()
  File "/home/ubuntu2004/isaac-sim4.5/IsaacLab/source/isaaclab/isaaclab/app/app_launcher.py", line 642, in _create_app
    self._app = SimulationApp(self._sim_app_config, experience=self._sim_experience_file)
  File "/home/ubuntu2004/isaac-sim3_25/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py", line 214, in __init__
    from .utils import create_new_stage, open_stage, set_livesync_stage
  File "/home/ubuntu2004/isaac-sim3_25/exts/omni.isaac.kit/omni/isaac/kit/utils.py", line 17, in <module>
    from omni.kit.usd import layers
ModuleNotFoundError: No module named 'omni.kit.usd'
2025-04-05 15:43:37 [5,398ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
o robot learning, that are not bugs or feature requests will be closed without reply, because GitHub issues are not an appropriate venue for these.

Advanced/nontrivial questions, especially in areas where documentation is lacking, are very much welcome.

For questions that are related to running and understanding Isaac Sim, please post them at the official [Isaac Sim forums](https://docs.omniverse.nvidia.com/app_isaacsim/app_isaacsim/isaac_sim_forums.html).

agj-neu, 2025-04-06:
absl-py                              2.2.2
action-msgs                          1.2.1
action-tutorials-interfaces          0.20.5
action-tutorials-py                  0.20.5
actionlib-msgs                       4.2.4
addict                               2.4.0
aiobotocore                          2.12.1
aiodns                               3.1.1
aiofiles                             23.2.1
aiohappyeyeballs                     2.6.1
aiohttp                              3.9.4
aiohttp-cors                         0.7.0
aioitertools                         0.11.0
aiosignal                            1.3.1
ament-cmake-test                     1.3.11
ament-copyright                      0.12.11
ament-cppcheck                       0.12.11
ament-cpplint                        0.12.11
ament-flake8                         0.12.11
ament-index-python                   1.4.0
ament-lint                           0.12.11
ament-lint-cmake                     0.12.11
ament-package                        0.14.0
ament-pep257                         0.12.11
ament-uncrustify                     0.12.11
ament-xmllint                        0.12.11
angles                               1.15.0
annotated-types                      0.7.0
antlr4-python3-runtime               4.9.3
anyio                                4.4.0
argcomplete                          3.6.0
asteval                              0.9.32
asttokens                            3.0.0
async-timeout                        4.0.3
attrs                                23.2.0
azure-core                           1.28.0
azure-identity                       1.13.0
azure-storage-blob                   12.17.0
blinker                              1.9.0
boto3                                1.26.63
botocore                             1.34.68
builtin-interfaces                   1.2.1
cchardet                             2.1.7
certifi                              2024.7.4
cffi                                 1.17.0
charset-normalizer                   3.3.2
click                                8.1.7
cloudpickle                          3.1.1
coloredlogs                          15.0.1
comm                                 0.2.2
commentjson                          0.9.0
composition-interfaces               1.2.1
ConfigArgParse                       1.7
construct                            2.10.68
contourpy                            1.3.1
coverage                             7.4.4
cryptography                         42.0.7
cv-bridge                            3.2.1
cycler                               0.11.0
dash                                 3.0.0
decorator                            4.4.2
demo-nodes-py                        0.20.5
diagnostic-msgs                      4.2.4
distro                               1.9.0
docker-pycreds                       0.4.0
domain-coordinator                   0.10.0
egl-probe                            1.0.2
einops                               0.8.1
example-interfaces                   0.9.3
examples-rclpy-executors             0.15.3
[... 298 lines cut ...]
transformers                         4.26.1
trimesh                              4.6.6
triton                               3.1.0
turtlesim                            1.4.2
typing_extensions                    4.11.0
tzdata                               2025.1
unique-identifier-msgs               2.2.1
urllib3                              2.2.2
usd-core                             24.11
uvicorn                              0.29.0
visualization-msgs                   4.2.4
vuer                                 0.0.32rc7
wandb                                0.12.21
warp-lang                            1.7.0
watchdog                             4.0.0
waterbear                            2.6.8
wcwidth                              0.2.13
webbot                               0.34
websockets                           12.0
Werkzeug                             3.0.6
wheel                                0.45.1
widgetsnbextension                   4.0.13
wrapt                                1.16.0
yarl                                 1.9.4
zipp                                 3.21.0

RandomOakForest, 2025-04-07:
Thank you for posting this. A user posted [this workaround](https://github.com/isaac-sim/IsaacLab/issues/1937#issuecomment-2689798982). If this doesn't help you, could you please file a _Bug Report_ and fill in the requested info. Thanks.

closed 2025-04-07 by RandomOakForest