isaac-sim/IsaacLab #2257: [Question] Question title

Rwik2000, 2025-04-05:
I have a simple car model, made using isaacsim interface. Even when the wheel angular velocities are almost zero (i.e. the car is almost stationary), I get non zero torque on the wheel revolute joints along the axis. I presume the torque along the axis, when the car is stationary should be around 0 right?

```
        joint_wrenches = self.robot.root_physx_view.get_link_incoming_joint_force()  # (N, num_joints, 6)
        joint_torques_world = joint_wrenches[..., 3:]  # (N, num_joints, 3)
        wheel_torques_local = qrotate_torch(self.wheel_quat_world_inv.view(-1, 4), 
                                            joint_torques_world[:, self.wheel_joint_ids].view(-1, 3)).view_as(joint_torques_world[:, self.wheel_joint_ids])
```

I am getting the wheel joint forces and torques [in the world frame] and converting it to the wheel frame using `FrameTransformerCfg`. I am spawning the car such the the world X axis aligns with the wheel revolute joint axis. I can confirm that the torque in the world X axis and the torque in the body frame along the axis are almost the same and are non-zero i.e. in the range of 0.5-2 Nm even when the car is stationary.

Am I doing something wrong to obtain the torques? Is this is an issue with the Physx in isaacsim or isaaclab?

[image]

Rwik2000, 2025-04-06:
[image]

This is the collision mesh visualization and I see that actual object and collision mesh are too much mis-aligned. When I just load the usd in isaacsim, I dont see the meshes to be mis-aligned.

Here is how I am loading the the robot : 

```
        self.isaaclab_car = ArticulationCfg(
            spawn=sim_utils.UsdFileCfg(
                usd_path=self.car_usd_path,
                scale=[self.scale, self.scale, self.scale],
                rigid_props=sim_utils.RigidBodyPropertiesCfg(
                    rigid_body_enabled=True,
                    max_depenetration_velocity=1.0,
                    enable_gyroscopic_forces=False,  # Disable unwanted angular forces
                ),

            articulation_props=sim_utils.ArticulationRootPropertiesCfg(
                    solver_position_iteration_count=32,
                    solver_velocity_iteration_count=32,

                ),

            mass_props = MassPropertiesCfg(
                                mass=self.mass,
                ),

            ),

            init_state=ArticulationCfg.InitialStateCfg(

                pos=self.start[:3],
                rot=self.start[3:7],
                lin_vel=[0.0, 0.0, 0.0],  # Ensure no initial velocity
                ang_vel=[0.0, 0.0, 0.0],  # Ensure no initial angular velocity
                # joint_pos={
                #     "Front_Suspension": -0.15,
                #     "Rear_Suspension": -0.15,
                # },
            ),

            actuators={
                "WheelEffort": ImplicitActuatorCfg(
                    joint_names_expr=["Knuckle.*"],  # or your wheel joint pattern
                    stiffness=0.0,
                    damping=0.0,
                    effort_limit=10000.0,  # Must be > 1000
                ),

                "Front_Axle_Steer": ImplicitActuatorCfg(
                    joint_names_expr=["Front_Axle.*"], #
                    stiffness=1000000,
                    damping=0,  # Prevent unintentional movement
                    effort_limit=100000,
                    velocity_limit=1000.0,
                ),

            },
        ) 
```

Maybe I am loading it in the wrong way? Has anyone seen this issue. Maybe this is causing the non-zero torques?

Rwik2000, 2025-04-07:
I think my way of calculating the torque on a rigid body is wrong. Can anybody point me to the correct documentation which gives us a way to have the torque acting on individual rigid bodies?

RandomOakForest, 2025-04-07:
Thanks for posting this. Basically, the torques are not in the world frame, but in the _child joint_ frame. There is not transform needed. For example, you should be able to get the _x_-axis torque using:
```
 self.robot.root_physx_view.get_link_incoming_joint_force()[..., 0]
```
This is from the [PhysX SDK used by Isaac Lab](https://nvidia-omniverse.github.io/PhysX/physx/5.6.0/_api_build/classPxArticulationCache.html#_CPPv4N19PxArticulationCache22linkIncomingJointForceE).

RandomOakForest, 2025-04-08:
It is the torque transmitted from the parent link, such as the chassis, to the wheel, in the wheel frame. Here the wheel frame is the child frame. The net torque, since the wheel has a free DOF in the wheel's axis, will be the sum of the torques applied by the drive, plus any limits (e.g. velocity) or joint friction. This is the [doc with more details](https://nvidia-omniverse.github.io/PhysX/physx/5.6.0/docs/Articulations.html#link-incoming-joint-force).

Rwik2000, 2025-04-09:
This figure gives a really good explanation of what `get_link_incoming_joint_force` is actually.

Is there any way to get the other forces as well in isaaclab/isaacsim?

RandomOakForest, 2025-04-09:
Maybe we are not understanding what you mean by the net force. How do you see what is missing? If you could elaborate would be helpful.

Rwik2000, 2025-04-09:
[image] 

This is the diagram from the documentation.

I want net force and torque due to all of the sources and not just the incoming joint force.

[image]
and similarly for torques

RandomOakForest, 2025-04-11:
Have you considered using a [contact sensor](https://isaac-sim.github.io/IsaacLab/main/source/overview/core-concepts/sensors/contact_sensor.html)?

Rwik2000, 2025-04-11:
Yes I have tried using it. I realised it just gives me the normal forces. 

Okay this is helpful then, I will use `get_link_accelerations`

closed 2025-04-11 by RandomOakForest