isaac-sim/IsaacLab #2259: [Bug Report] Encounting a conda environment issue and could not launch isaaclab from it

jingyang-huang, 2025-04-06:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

A clear and concise description of what the bug is.

### Steps to reproduce

Follow the binary installation in https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/binaries_installation.html

Before **Setting up the conda environment (optional)** step, everything is fine, and I could run issac-sim with the following test:
```shell
# checks that python path is set correctly
${ISAACSIM_PYTHON_EXE} -c "print('Isaac Sim configuration is now complete.')"
# checks that Isaac Sim can be launched from python
${ISAACSIM_PYTHON_EXE} ${ISAACSIM_PATH}/standalone_examples/api/isaacsim.core.api/add_cubes.py
```
And I set up the link to isaac_sim
```shell
# enter the cloned repository
cd IsaacLab
# create a symbolic link
ln -s ${HOME}/isaacsim _isaac_sim
```
Then I try to install  all the isaaac extensions in source directory 
```shell
# in default python env
./isaaclab.sh --install # or "./isaaclab.sh -i"
```
I verify the installation using option 1 and it is fine:

```shell
# Option 1: Using the isaaclab.sh executable
# note: this works for both the bundled python and the virtual environment
./isaaclab.sh -p scripts/tutorials/00_sim/create_empty.py
```

Nextly I follow the steps in **Setting up the conda environment** ,
```shell
./isaaclab.sh --conda  isaaclab
```
When I try to activate the conda env, the issue occurs

```shell
hjy@hjy:~/HJY/Codes/ISSAC/IsaacLab$ conda activate isaaclab
Fatal Python error: init_sys_streams: can't initialize sys standard streams
Python runtime state: core initialized
Traceback (most recent call last):
  File "/home/hjy/HJY/Codes/ISSAC/IsaacLab/_isaac_sim/kit/python/lib/python3.10/io.py", line 54, in <module>
ImportError: cannot import name 'text_encoding' from 'io' (unknown location)
Fatal Python error: init_sys_streams: can't initialize sys standard streams
Python runtime state: core initialized
Traceback (most recent call last):
  File "/home/hjy/HJY/Codes/ISSAC/IsaacLab/_isaac_sim/kit/python/lib/python3.10/io.py", line 54, in <module>
ImportError: cannot import name 'text_encoding' from 'io' (unknown location)

```
And I could not pass the verifying step as well, the output is as follows:

```shell
(isaaclab) hjy@hjy:~/HJY/Codes/ISSAC/IsaacLab$ ./isaaclab.sh -p scripts/tutorials/00_sim/create_empty.py
[INFO] Using python from: /home/hjy/anaconda3/envs/isaaclab/bin/python                                                
Traceback (most recent call last):
  File "/home/hjy/HJY/Codes/ISSAC/IsaacLab/scripts/tutorials/00_sim/create_empty.py", line 20, in <module>
    from isaaclab.app import AppLauncher
ModuleNotFoundError: No module named 'isaaclab'
(isaaclab) hjy@hjy:~/HJY/Codes/ISSAC/IsaacLab$ python scripts/tutorials/00_sim/create_empty.py
Traceback (most recent call last):
  File "/home/hjy/HJY/Codes/ISSAC/IsaacLab/scripts/tutorials/00_sim/create_empty.py", line 20, in <module>
    from isaaclab.app import AppLauncher
[... 111 lines cut ...]
bash: /home/hjy/anaconda3/envs/isaaclab/bin:/home/hjy/.dotnet:/opt/ros/noetic/bin:/home/hjy/Downloads/Libs/TensorRT-8.5.2.2/bin:/home/hjy/.dotnet:/home/hjy/anaconda3/condabin:/home/hjy/Downloads/Libs/TensorRT-8.5.2.2/bin:/home/hjy/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/usr/local/cuda/bin:/usr/local/cuda/bin: No such file or directory
(isaaclab) hjy@hjy:~/HJY/Codes/ISSAC/IsaacLab$ $PYTHONPATH
```
For the sake of contrast, I output the same in the default python env

```shell
hjy@hjy:~/HJY/Codes/ISSAC/IsaacLab$ env | grep -i python
ROS_PYTHON_VERSION=3
ISAACSIM_PYTHON_EXE=/home/hjy/isaacsim/python.sh
PYTHONPATH=/opt/ros/noetic/lib/python3/dist-packages
CONDA_PYTHON_EXE=/home/hjy/anaconda3/bin/python
hjy@hjy:~/HJY/Codes/ISSAC/IsaacLab$ which pip
/home/hjy/.local/bin/pip
hjy@hjy:~/HJY/Codes/ISSAC/IsaacLab$ which python
/usr/bin/python
hjy@hjy:~/HJY/Codes/ISSAC/IsaacLab$ $PATH
bash: /home/hjy/.dotnet:/opt/ros/noetic/bin:/home/hjy/anaconda3/condabin:/home/hjy/Downloads/Libs/TensorRT-8.5.2.2/bin:/home/hjy/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/usr/local/cuda/bin: No such file or directory
hjy@hjy:~/HJY/Codes/ISSAC/IsaacLab$ $PYTHONPATH
bash: /opt/ros/noetic/lib/python3/dist-packages: Is a directory
```

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

RandomOakForest, 2025-04-07:
Thank you for posting this. Have you tried verifying your installation and activating your conda environment after restarting your workstation? If you had earlier versions of Isaac Sim, you may have to use `--reset-user`. [See note here](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/binaries_installation.html). Let us know if you need further help.

jingyang-huang, 2025-04-08:
> Thank you for posting this. Have you tried verifying your installation and activating your conda environment after restarting your workstation? If you had earlier versions of Isaac Sim, you may have [...]

Thank you for replying. I have tried to verify my isaac sim and isaac lab installation after rebooting my workstation, and they works fine with the command:
```shell
./isaaclab.sh --sim
./isaaclab.sh -p scripts/tutorials/00_sim/create_empty.py
```
However, the same does not work well when I switch to the conda environment. And I notice every time I activate the conda environment, the error` ImportError: cannot import name 'text_encoding' from 'io' (unknown location)` occurs. But once I comment the `setup_conda_env.sh` , the ImportError disappears, and If I mannualy source the `setup_conda_env.sh`, the error show again. So I guess it was caused by this bash. 
```shell
#!/usr/bin/env bash

# for Isaac Lab
export ISAACLAB_PATH=/home/hjy/HJY/Codes/ISSAC/IsaacLab
alias isaaclab=/home/hjy/HJY/Codes/ISSAC/IsaacLab/isaaclab.sh

# show icon if not runninng headless
export RESOURCE_NAME="IsaacSim"

# for Isaac Sim
# source /home/hjy/HJY/Codes/ISSAC/IsaacLab/_isaac_sim/setup_conda_env.sh
```
However, even if I did not source the `setup_conda_env.sh` and avoid the ImportError, I stil could not run the command `./isaaclab.sh -p scripts/tutorials/00_sim/create_empty.py` with the same error: `ModuleNotFoundError: No module named 'isaaclab' `
Besides, I was using a older version isaac sim 4.1 before. I noticed the notation in the installation steps and I have added the external args `--reset-user` when I launch my isaac sim for the first time, but the error remains.
Additionally, I have asked the same question with one my friend who has installed the same version on one brand new workstation, it seems he did not meet such problem and his conda environment works fine. So I guess it was due to some conflict with older versions, and related to ` source /home/hjy/HJY/Codes/ISSAC/IsaacLab/_isaac_sim/setup_conda_env.sh`

jingyang-huang, 2025-04-22:
Hi, everyone, I`m very happy to tell you that I fix this problem. 
The cause is that multi python env in conda would cause isaac python env not works
So my solution is uninstall Anaconda and remove all envs, and reinstall Anaconda. Then I started to create isaac conda env first, and this is my conda env result.
```shell
hjy@hjy:~/HJY/Codes/ISSAC/IsaacLab$ conda env list
# conda environments:
#
base                     /home/hjy/anaconda3
isaaclab                 /home/hjy/anaconda3/envs/isaaclab
```
May my solution help you.

I8Robot, 2025-05-13:
> Hi, everyone, I am very happy to tell you that I fix this problem. The cause is that multi python env in conda would cause isaac python env not works So my solution is uninstall Anaconda (`rm -rf ~/anaconda3`) and remove all envs, and reinstall Anaconda. Then I started to create isaac conda env first, and this is my conda env result.
> 
> hjy@hjy:~/HJY/Codes/ISSAC/IsaacLab$ conda env list
> # conda environments:
> #
> base                     /home/hjy/anaconda3
> isaaclab                 /home/hjy/anaconda3/envs/isaaclab
> May my solution help you.

There can only isaac conda env?

jingyang-huang, 2025-05-13:
> There can only isaac conda env?

Hi iRobot, I fix this problem and update my comment , pls check it.
My answer is that there can be multiple python conda env using my solution