isaac-sim/IsaacLab #2262: Cpu 100% when launching isaacsim

Mahuuu0408, 2025-04-06:
This is the log file:

Loading user config located at: ‘C:/Users/Administrator/AppData/Local/ov/data/Kit/Isaac-Sim Full/4.5/user.config.json’
[Info] [carb] Logging to file: C:/Users/Administrator/.nvidia-omniverse/logs/Kit/Isaac-Sim Full/4.5/kit_20250406_165036.log
[0.230s] [ext: omni.kit.async_engine-0.0.1] startup
[0.672s] [ext: omni.metrics.core-0.0.1] startup
[0.675s] [ext: omni.client.lib-1.0.0] startup
[0.693s] [ext: omni.blobkey-1.1.2] startup
[0.695s] [ext: omni.stats-1.0.1] startup
[0.698s] [ext: omni.datastore-0.0.0] startup
[0.700s] [ext: omni.client-1.2.2] startup
[0.712s] [ext: omni.ujitso.default-1.0.0] startup
[0.716s] [ext: omni.hsscclient-1.1.1] startup
2025-04-06 15:50:37 [663ms] [Warning] [omni.datastore] OmniHub is inaccessible
[0.722s] [ext: omni.gpu_foundation.shadercache.vulkan-1.0.0] startup
[0.729s] [ext: omni.assets.plugins-0.0.0] startup
[0.733s] [ext: omni.gpu_foundation-0.0.0] startup
[0.763s] [ext: carb.windowing.plugins-1.0.0] startup
[0.954s] [ext: omni.kit.renderer.init-0.0.0] startup
[0.957s] [ext: omni.kit.pipapi-0.0.0] startup
[0.975s] [ext: omni.kit.pip_archive-0.0.0] startup
[0.977s] [ext: omni.pip.compute-1.4.5] startup
[0.979s] [ext: omni.pip.cloud-1.1.7] startup
[0.984s] [ext: omni.isaac.core_archive-2.3.3] startup
[0.986s] [ext: omni.materialx.libs-1.0.6] startup
[1.019s] [ext: omni.isaac.ml_archive-2.1.2] startup
[1.021s] [ext: omni.usd.config-1.0.5] startup
[1.054s] [ext: omni.gpucompute.plugins-0.0.0] startup
[1.059s] [ext: omni.usd.libs-1.0.1] startup
[1.198s] [ext: omni.kit.telemetry-0.5.1] startup
[1.220s] [ext: omni.mdl-55.0.1] startup
[1.255s] [ext: omni.iray.libs-0.0.0] startup
[1.321s] [ext: omni.mdl.neuraylib-0.2.10] startup
[1.329s] [ext: omni.kit.usd.mdl-1.0.7] startup
2025-04-06 15:50:38 [1,351ms] [Warning] [gpu.foundation.plugin] Skipping unsupported non-NVIDIA GPU: Intel(R) RaptorLake-S Mobile Graphics Controller
2025-04-06 15:50:38 [1,352ms] [Warning] [gpu.foundation.plugin] Skipping unsupported non-NVIDIA GPU: Intel(R) RaptorLake-S Mobile Graphics Controller
[1.494s] [ext: omni.kit.loop-isaac-1.2.1] startup
[1.499s] [ext: omni.kit.test-1.1.2] startup
[1.614s] [ext: omni.appwindow-1.1.9] startup
[1.636s] [ext: omni.kit.renderer.core-1.0.2] startup
[1.646s] [ext: omni.kit.renderer.capture-0.0.0] startup
[1.656s] [ext: omni.kit.renderer.imgui-1.0.2] startup

|---------------------------------------------------------------------------------------------|
| Driver Version: 572.83 | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name | Active | LDA | GPU Memory | Vendor-ID | LUID |
| | | | | | Device-ID | UUID |
| | | | | | Bus-ID | |
|---------------------------------------------------------------------------------------------|
| 0 | NVIDIA GeForce RTX 4060 Laptop.. | Yes: 0 | | 7957 MB | 10de | dbea0000.. |
| | | | | | 28a0 | dc908023.. |
| | | | | | 1 | |
|---------------------------------------------------------------------------------------------|
| 1 | Intel(R) RaptorLake-S Mobile G.. | | | 16248 MB | 8086 | fbe60000.. |
| | | | | | a78b | 86808ba7.. |
| | | | | | N/A | |
|=============================================================================================|
| OS: Windows 11 Pro for Workstations, Version: 10.0 (24H2), Build: 26100, Kernel: 10.0.26100.3624
| Processor: Intel(R) Core™ i7-14650HX
| Cores: 16 | Logical Cores: 24
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 32496 | Free Memory: 20262
| Total Page/Swap (MB): 37616 | Free Page/Swap: 23147
|---------------------------------------------------------------------------------------------|
[1.679s] [ext: omni.ui-2.26.5] startup
[1.747s] [ext: omni.kit.mainwindow-1.0.3] startup
[1.752s] [ext: carb.audio-0.1.0] startup
[1.755s] [ext: omni.uiaudio-1.0.0] startup
[1.760s] [ext: omni.kit.uiapp-0.0.0] startup
[... 505 lines cut ...]
2025-04-06 15:50:46 [9,747ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit : valid true, within: true
2025-04-06 15:50:49 [12,192ms] [Warning] [omni.usd-abi.plugin] No setting was found for ‘/rtx-defaults-transient/meshlights/forceDisable’
2025-04-06 15:50:49 [12,235ms] [Warning] [omni.usd-abi.plugin] No setting was found for ‘/rtx-defaults/post/dlss/execMode’

Isaac Sim Version
 4.5.0

Operating System
 Windows 11

GPU Information
Model: NVIDIA GeForce RTX 4060 Laptop
Driver Version: 572.83
Topic Description
Detailed Description
just try to open isaacsim, the cpu is 100% used and isaacsim is non-responsed (as seen in the screenshot).

[image]

[image]

Addition info
I have tried this way, I changed the ‘‘false’’ to ‘‘true’’ and the rest two lines unchanged, but it did not work.

[image]

RandomOakForest, 2025-04-07:
Thank you for posting this. Is your Isaac Sim crashing? Have you tried [these speed-up suggestions](https://docs.omniverse.nvidia.com/isaacsim/latest/reference_material/speedup_cheat_sheet.html)?

Ethereal1679, 2025-04-23:
Hi, unfortunately, I got the same problems:
```
2025-04-23 11:48:50 [9,736ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults-transient/meshlights/forceDisable'
2025-04-23 11:48:50 [9,776ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/post/dlss/execMode'
```
Have u solved it ?????

Derman0524, 2025-04-25:
> Hi, unfortunately, I got the same problems:
>
> ```
> 2025-04-23 11:48:50 [9,736ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults-transient/meshlights/forceDisable'
> 2025-04-23 [...]

Same issue with me! I got a brand new 5090 the other day and I can't run Isaac Lab, it gets stuck on the last line. Even in Headless mode I get issues. Not sure what to do

juliushaag, 2025-04-26:
Same for me, isaac sim takes ~5min with 100% CPU utilisation (close to no network activity) until its usable.
If you use the verbose flag during console start up it shows some progress but it is happening every time i start isaac sim.

RandomOakForest, 2025-05-10:
Following up on this, it could be that your GPU driver is incorrectly installed. Could you reinstall it and try again? If you have issues with a 50 series GPU, please open a new issue. Thanks.

closed 2025-05-10 by RandomOakForest