isaac-sim/IsaacLab #2274: [Question] Question title : Unable to find 'omniverse-kit' package

MK040412, 2025-04-09:
### Question

I'm facing an error that "Unable to find 'omniverse-kit' package"

(isaaclab) user@user-desktop:~/IsaacLab$ python scripts/tutorials/00_sim/create_empty.py
Unable to find 'omniverse-kit' package

However, in fact, pip list shows to me omniverse-kit lib

omniverse-kit              106.5.0.162521

Why I'm facing with this problem I'm trying Isaac sim 4.5.0 and IsaacLab 2.0.0

My desktop is Ubuntu 22.04 desktop and there is no error log during the setup

What is the  problem with setup?

Sincerely

Below is my pip list

Package                    Version        Editable project location
-------------------------- -------------- ----------------------------------------------
absl-py                    2.2.2
aiowsgi                    0.8
antlr4-python3-runtime     4.9.3
anyio                      4.9.0
argon2-cffi                23.1.0
argon2-cffi-bindings       21.2.0
arrow                      1.3.0
asttokens                  3.0.0
async-lru                  2.0.5
attrs                      25.3.0
babel                      2.17.0
beautifulsoup4             4.13.3
bleach                     6.2.0
boto3                      1.37.30
botocore                   1.37.30
certifi                    2025.1.31
cffi                       1.17.1
charset-normalizer         3.4.1
click                      8.1.8
cloudpickle                3.1.1
comm                       0.2.2
contourpy                  1.3.1
cycler                     0.12.1
debugpy                    1.8.13
decorator                  4.4.2
defusedxml                 0.7.1
docker-pycreds             0.4.0
docopt                     0.6.2
egl_probe                  1.0.2
einops                     0.8.1
exceptiongroup             1.2.2
executing                  2.2.0
Farama-Notifications       0.0.4
fastjsonschema             2.21.1
filelock                   3.18.0
fonttools                  4.57.0
fqdn                       1.5.1
fsspec                     2025.3.2
gitdb                      4.0.12
GitPython                  3.1.44
grpcio                     1.71.0
gym                        0.23.1
gym-notices                0.0.8
gymnasium                  1.1.1
h11                        0.14.0
h5py                       3.13.0
hidapi                     0.14.0.post2
httpcore                   1.0.7
httpx                      0.28.1
huggingface-hub            0.30.2
hydra-core                 1.3.2
idna                       3.10
imageio                    2.37.0
imageio-ffmpeg             0.6.0
ipykernel                  6.29.5
ipython                    8.34.0
ipywidgets                 8.1.5
isaaclab                   2.0.2
isaaclab_assets            0.2.1          /home/perelman/IsaacLab/source/isaaclab_assets
isaaclab_mimic             1.0.3          /home/perelman/IsaacLab/source/isaaclab_mimic
isaaclab_rl                0.1.3          /home/perelman/IsaacLab/source/isaaclab_rl
isaaclab_tasks             0.10.27        /home/perelman/IsaacLab/source/isaaclab_tasks
isaacsim                   4.5.0.0
isaacsim-app               4.5.0.0
isaacsim-asset             4.5.0.0
isaacsim-benchmark         4.5.0.0
isaacsim-code-editor       4.5.0.0
isaacsim-core              4.5.0.0
isaacsim-cortex            4.5.0.0
isaacsim-example           4.5.0.0
isaacsim-extscache-kit     4.5.0.0
isaacsim-extscache-kit-sdk 4.5.0.0
isaacsim-extscache-physics 4.5.0.0
isaacsim-gui               4.5.0.0
isaacsim-kernel            4.5.0.0
isaacsim-replicator        4.5.0.0
isaacsim-rl                4.5.0.0
isaacsim-robot             4.5.0.0
isaacsim-robot-motion      4.5.0.0
isaacsim-robot-setup       4.5.0.0
isaacsim-ros1              4.5.0.0
isaacsim-ros2              4.5.0.0
isaacsim-sensor            4.5.0.0
isaacsim-storage           4.5.0.0
isaacsim-template          4.5.0.0
isaacsim-test              4.5.0.0
isaacsim-utils             4.5.0.0
isoduration                20.11.0
jedi                       0.19.2
Jinja2                     3.1.6
jmespath                   1.0.1
json5                      0.10.0
jsonpointer                3.0.0
jsonschema                 4.23.0
jsonschema-specifications  2024.10.1
jupyter_client             8.6.3
jupyter_core               5.7.2
jupyter-events             0.12.0
jupyter-lsp                2.2.5
jupyter_server             2.15.0
jupyter_server_terminals   0.5.3
jupyterlab                 4.3.6
jupyterlab_pygments        0.3.0
jupyterlab_server          2.27.3
jupyterlab_widgets         3.0.13
kiwisolver                 1.4.8
Markdown                   3.7
MarkupSafe                 3.0.2
matplotlib                 3.10.1
matplotlib-inline          0.1.7
mistune                    3.1.3
moviepy                    1.0.3
mpmath                     1.3.0
nbclient                   0.10.2
nbconvert                  7.16.6
nbformat                   5.10.4
nest-asyncio               1.6.0
networkx                   3.4.2
notebook                   7.3.3
notebook_shim              0.2.4
numpy                      1.26.4
nvidia-cublas-cu12         12.4.5.8
nvidia-cuda-cupti-cu12     12.4.127
nvidia-cuda-nvrtc-cu12     12.4.127
nvidia-cuda-runtime-cu12   12.4.127
nvidia-cudnn-cu12          9.1.0.70
nvidia-cufft-cu12          11.2.1.3
nvidia-curand-cu12         10.3.5.147
nvidia-cusolver-cu12       11.6.1.9
nvidia-cusparse-cu12       12.3.1.170
nvidia-nccl-cu12           2.21.5
nvidia-nvjitlink-cu12      12.4.127
nvidia-nvtx-cu12           12.4.127
omegaconf                  2.3.0
omni                       0.0.2
omniverse-kit              106.5.0.162521
onnx                       1.16.1
opencv-python              4.11.0.86
osqp                       0.6.7.post3
overrides                  7.7.0
packaging                  24.2
pandas                     2.2.3
pandocfilters              1.5.1
parso                      0.8.4
pathtools                  0.1.2
pexpect                    4.9.0
pillow                     11.0.0
pip                        25.0.1
platformdirs               4.3.7
prettytable                3.3.0
proglog                    0.1.11
prometheus_client          0.21.1
promise                    2.3
prompt_toolkit             3.0.50
protobuf                   3.20.3
psutil                     5.9.8
ptyprocess                 0.7.0
pure_eval                  0.2.3
pycparser                  2.22
pygame                     2.1.0
pyglet                     1.5.31
Pygments                   2.19.1
pyparsing                  3.2.3
pyperclip                  1.9.0
python-dateutil            2.9.0.post0
python-json-logger         3.3.0
pytz                       2025.2
PyYAML                     6.0.2
pyzmq                      26.4.0
qdldl                      0.1.7.post5
referencing                0.36.2
regex                      2024.11.6
requests                   2.32.3
rfc3339-validator          0.1.4
rfc3986-validator          0.1.1
rl_games                   1.6.1
robomimic                  0.4.0
rpds-py                    0.23.1
rsl-rl-lib                 2.3.0
s3transfer                 0.11.4
safetensors                0.5.3
schema                     0.7.7
scipy                      1.15.2
seaborn                    0.13.2
Send2Trash                 1.8.3
sentry-sdk                 2.25.1
setproctitle               1.3.5
setuptools                 75.8.0
shortuuid                  1.0.13
six                        1.17.0
skrl                       1.4.3
smmap                      5.0.2
sniffio                    1.3.1
soupsieve                  2.6
stable_baselines3          2.6.0
stack-data                 0.6.3
starlette                  0.46.0
sympy                      1.13.1
tensorboard                2.19.0
tensorboard-data-server    0.7.2
tensorboardX               2.6.2.2
termcolor                  3.0.1
terminado                  0.18.1
tinycss2                   1.4.0
tokenizers                 0.21.1
toml                       0.10.2
tomli                      2.2.1
torch                      2.5.1
torchvision                0.20.1
tornado                    6.4.2
tqdm                       4.67.1
traitlets                  5.14.3
transformers               4.51.1
trimesh                    4.6.6
triton                     3.1.0
types-python-dateutil      2.9.0.20241206
typing_extensions          4.13.1
tzdata                     2025.2
uri-template               1.3.0
urllib3                    2.3.0
waitress                   3.0.2
wandb                      0.12.21
warp-lang                  1.7.0
wcfg                       6
wcwidth                    0.2.13
webcolors                  24.11.1
webencodings               0.5.1
WebOb                      1.8.9
websocket-client           1.8.0
Werkzeug                   3.1.3
wheel                      0.45.1
widgetsnbextension         4.0.13

RandomOakForest, 2025-04-09:
Thanks for posting this. Some users have hit this before, have you tried using venv instead of conda?

Toni-SM, 2025-04-09:
Hi @MK040412 

The problem is the [omni](https://pypi.org/project/omni/) package (`omni 0.0.2`) that has nothing to do with NVIDIA Omniverse [omniverse-kit](https://pypi.org/project/omniverse-kit/) package required by Isaac Sim/Lab.
 
The `omniverse-kit` package (installed as the `omni` folder) was overwritten by the other `omni` package.

Do fresh install in a new (virtual) Python environment to solve it.

MK040412, 2025-04-09:
@Toni-SM  You are my life saver, THANK YOU SO A LOT
And ALSO @RandomOakForest  You are my life saver, thank you.

just do a one line and it works.

```bash

pip uninstall omni

```

I used almost 4 days, day and night, struggle with this problem, thank you.

MK040412, 2025-04-09:
- [ ] 

- [ ]

closed 2025-04-09 by MK040412