isaac-sim/IsaacLab #2275: [Proposal] Demo environments for Model Predictive Control Tasks

Zuwa92, 2025-04-09:
### Proposal

It would be beneficial if the IsaacLab repo includes a demo that demonstrates the use of Isaaclab for model predictive control. A good example would be an implementation of a gradient-free (sampling-based) path /motion optimization example. 
### Motivation

I think this will be valuable because it can give users a good starting point (or reference) for implementing similar frameworks. Also, the sampling-based algorithms would be greatly facilitated by the parallelism Isaaclab offers and can serve as an alternative to RL.

### Alternatives

N/A

### Additional context

N/A

### Checklist

- [ X ] I have checked that there is no similar issue in the repo (**required**)

Tadinu, 2025-06-06:
Hi, these may be relevant, as various implementations of MPPI using torch
https://autonomousrobots.nl/paper_websites/isaac-mppi -> https://github.com/tud-amr/mppi_torch
https://stochasticmppi.github.io -> https://github.com/RIVeR-Lab/Stochastic-MPPI
https://sites.google.com/view/residual-mppi -> https://github.com/outsider86/Residual-MPPI
https://github.com/UM-ARM-Lab/pytorch_mppi
https://github.com/kohonda/mppi_playground
https://github.com/pytorch/rl/blob/main/torchrl/modules/planners/mppi.py
...

Also, Curobo has one implemented in IsaacSim
https://curobo.org/get_started/2b_isaacsim_examples.html#model-predictive-control-with-mppi
https://github.com/NVlabs/curobo/discussions/441
https://github.com/NVlabs/curobo/discussions/375
, so would be nice to have it integrated into IsaacLab env

For params
https://github.com/ros-navigation/navigation2/tree/main/nav2_mppi_controller

AntoineRichard, 2026-08-27:
@Zuwa92 thanks for the proposal, and @Tadinu for the pointers to existing work.

We're going to leave sampling-based MPC out of the core repo for now — there's no sampling-based MPC/MPPI implementation under `source/` or `scripts/` (the OCS2 MPC named in the motion-generators doc is an external planner we wrap, not in-tree), and the ecosystem already covers it well on top of Isaac Sim/Isaac Lab parallelism: cuRobo's Isaac Sim MPPI example, `pytorch_mppi`, and `mppi_torch`.

That said, if you or anyone else wants to contribute a self-contained demo under `scripts/demos`, open a PR and we'll gladly review it. Closing for now.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard