isaac-sim/IsaacLab #2276: [Question] Mobile franka init pose error

saiyao233, 2025-04-09:
[image]

[image]

I wanna know why the robot is initialized in this pose, like the image shows and how can I fix it

RandomOakForest, 2025-04-09:
Thank you for posting this. Which versions of Isaac Lab and Isaac Sim are you using?

saiyao233, 2025-04-10:
> Thank you for posting this. Which versions of Isaac Lab and Isaac Sim are you using?
Isaaclab 0.36.5 and Isaacsim 4.5.0

saiyao233, 2025-04-11:
> Could you try updating to Isaac Lab 2.0.2? Thanks.
Sorry for my mistake; I think it is 2.0.2. I pulled the code from the main branch

closed 2025-04-15 by saiyao233