isaac-sim/IsaacLab #2280: "Combined robot simulation problem"

agj-neu, 2025-04-09:
I followed the provided Assemble tutorial to use a quadruped robot as a mobile base for a robotic arm, but when both the quadruped and the arm are in the same directory, Isaac Lab reports that it cannot find the arm joints during simulation.Below are my simulation environment and error message.
    # cartpole
    robot: ArticulationCfg = ArticulationCfg(
        prim_path="/World/envs/env_.*/Robot",
        spawn=sim_utils.UsdFileCfg(
            usd_path="/home/ubuntu2004/GRScenes/robots/aliengo/aliengo_camera_only.usd",
            rigid_props=sim_utils.RigidBodyPropertiesCfg(
                rigid_body_enabled=True,
                max_linear_velocity=1000.0,
                max_angular_velocity=1000.0,
                max_depenetration_velocity=100.0,
                enable_gyroscopic_forces=True,
            ),
            articulation_props=sim_utils.ArticulationRootPropertiesCfg(
                enabled_self_collisions=False,
                solver_position_iteration_count=4,
                solver_velocity_iteration_count=0,
                sleep_threshold=0.005,
                stabilization_threshold=0.001,
            ),
        ),
        init_state=ArticulationCfg.InitialStateCfg(
            pos=(0.0, 0.0, 1.5), 
            rot=(0.7071, -0.7071, 0.0, 0.0),
            #rot=(1.0, 0.0, 0.0, 0.0),  # (w,x,y,z)格式的四元数，表示直立姿态
            joint_pos={
                "FL_hip_joint": 0.0, "FR_hip_joint": 0.0,
                "RL_hip_joint": 0.0, "RR_hip_joint": 0.0,
                "FL_thigh_joint": 0.0, "FR_thigh_joint": 0.0,
                "RL_thigh_joint": 0.0, "RR_thigh_joint": 0.0,
                "FL_calf_joint": -1, "FR_calf_joint": -1,
                "RL_calf_joint": -1, "RR_calf_joint": -1,
                "panda_joint1": 0.0, "panda_joint2": 0.0,
                "panda_joint3": 0.0, "panda_joint4": -1,
                "panda_joint5": 0.0, "panda_joint6": 0.0,
                "panda_joint7": 0.0,
                #"panda_finger_joint1": 0.0, "panda_finger_joint2": 0.0, # 如需控制夹爪则取消注释
            },
        ),
        actuators={
            # 髋关节执行器 (hip joints)
            "fl_hip_actuator": ImplicitActuatorCfg(
                joint_names_expr=["FL_hip_joint"],
                effort_limit=400.0,
                velocity_limit=100.0,
                stiffness=40.0,    # 使用Aliengo中定义的kps值
                damping=2.0,       # 使用Aliengo中定义的kds值
            ),
            "fr_hip_actuator": ImplicitActuatorCfg(
                joint_names_expr=["FR_hip_joint"],
                effort_limit=400.0,
                velocity_limit=100.0,
                stiffness=40.0,
                damping=2.0,
            ),
            "rl_hip_actuator": ImplicitActuatorCfg(
                joint_names_expr=["RL_hip_joint"],
                effort_limit=400.0,
                velocity_limit=100.0,
                stiffness=40.0,
                damping=2.0,
            ),
            "rr_hip_actuator": ImplicitActuatorCfg(
                joint_names_expr=["RR_hip_joint"],
                effort_limit=400.0,
                velocity_limit=100.0,
                stiffness=40.0,
                damping=2.0,
            ),

[... 166 lines cut ...]
ValueError: Not all regular expressions are matched! Please check that the regular expressions are correct: 
	FL_hip_joint: ['FL_hip_joint']
	FR_hip_joint: ['FR_hip_joint']
	RL_hip_joint: ['RL_hip_joint']
	RR_hip_joint: ['RR_hip_joint']
	FL_thigh_joint: ['FL_thigh_joint']
	FR_thigh_joint: ['FR_thigh_joint']
	RL_thigh_joint: ['RL_thigh_joint']
	RR_thigh_joint: ['RR_thigh_joint']
	FL_calf_joint: ['FL_calf_joint']
	FR_calf_joint: ['FR_calf_joint']
	RL_calf_joint: ['RL_calf_joint']
	RR_calf_joint: ['RR_calf_joint']
	panda_joint1: []
	panda_joint2: []
	panda_joint3: []
	panda_joint4: []
	panda_joint5: []
	panda_joint6: []
	panda_joint7: []
Available strings: ['FL_hip_joint', 'FR_hip_joint', 'RL_hip_joint', 'RR_hip_joint', 'FL_thigh_joint', 'FR_thigh_joint', 'RL_thigh_joint', 'RR_thigh_joint', 'FL_calf_joint', 'FR_calf_joint', 'RL_calf_joint', 'RR_calf_joint']

"But when I placed the robotic arm as a subdirectory under the quadruped robot, the arm's joints were detected but couldn't move properly. The connection point seems to be fixed in place."​​

[image]

agj-neu, 2025-04-09:
[image]Another possible reason is rot=(0.7071, -0.7071, 0.0, 0.0)—I added this rotation because the loaded robot wasn't standing upright initially but was instead lying on its side. If possible, please tell me how to set my robot's initial pose to a standing position.

Bardreamaster, 2025-04-10:
@agj-neu Please use the ~~fucking~~ code backticks to make your codes clear, like this:

```python
a_python_func()
```