isaac-sim/IsaacLab #2281: [Proposal] Add compliant stiffness and damping randomization when randomizing materials

ParlitsisG, 2025-04-09:
### Proposal

Add compliant stiffness and damping randomization when randomizing materials. Currently physics material randomization only supports friction and restitution.

### Motivation
Manipulation tasks such as a manipulator pushing picking require randomizations in the behaviour of the object during contact to simulate the diversity of object behaviour during impact and contacts a compliant contact model which is randomized could include a good method for domain randomization.

### Alternatives
I guess using Deformable objects is possible but at much higher computational cost.

### Additional context
I am currently working on a project using a dual arm manipulation franka manipulator with flat siliconde end effectors instead of a gripper to pick and place large objects. In order to describe contacts I want to randomize the contact model as part of domain randomization to make my RL agent more robust during the Sim-to-real.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)

RandomOakForest, 2025-04-10:
Thank you for posting this proposal. Could you elaborate on what you mean by "compliant contact model" and give an example or elaborate in what would be a good implementation? Thanks.

RandomOakForest, 2025-04-11:
Thanks. The team has taken note of this.

Lynn-hh, 2026-09-23:
I'd like to take this. My plan:

1. **API.** Add optional `compliant_contact_stiffness_range` (log-uniform) and `compliant_contact_damping_range` to `randomize_rigid_body_material`. `None` keeps the current values.
2. **PhysX.** Read and write through `get/set_compliant_material_properties`, so friction DR no longer silently erases spawn-authored compliance. The erasure happens because `set_material_properties` writes damping = 0 and a non-negative restitution, and PhysX encodes compliant stiffness as negative restitution. This is what #7511 hit on KukaAllegro lift.
3. **Newton.** Write `shape_material_ke/kd` per env and shape, then notify `SHAPE_PROPERTIES`. SolverMuJoCo then recomputes `geom_solref`.
4. **OVPhysX.** Raise a clear `NotImplementedError` for now.
5. **Tests.** Pure-torch sampling tests, plus a GPU check (penetration depth vs stiffness on a compliant plate).

With this in place, #7511's task-local `RandomizeFrictionKeepCompliant` becomes a config line.

@ooctipus: should this target the current `envs/mdp/events.py`, or the per-backend `physics_events` layout from #5915?
@matthewtrepte: this came to you in the recent triage. OK if I pick it up?

Context: I work on material-aware peg-in-hole on a real Franka FR3, where contact stiffness DR is central. I can validate on H100s.