isaac-sim/IsaacLab #2286: [Question] number of physics material exceed 64K

xindonglin99, 2025-04-10:
### What's the correct way to spawn multiple assets with different prim path?

I'm trying to spawn 20 cubes in each environment as stepping stones. This is the code that I'm using
```
def create_step_cfg(num_steps: int, size: Tuple[float, float, float]) -> RigidObjectCollectionCfg:
    """Create a step configuration."""
    collection = {}
    spawn_cfg = sim_utils.CuboidCfg(
        size=size,
        rigid_props=sim_utils.RigidBodyPropertiesCfg(kinematic_enabled=True),
        collision_props=sim_utils.CollisionPropertiesCfg(collision_enabled=True),
        visual_material=sim_utils.PreviewSurfaceCfg(diffuse_color=(0.1, 0.1, 0.1), metallic=0.2),
        physics_material=sim_utils.RigidBodyMaterialCfg(
        friction_combine_mode="average",
        restitution_combine_mode="average",
        static_friction=1.0,
        dynamic_friction=1.0,
        restitution=0.0,
        ),
    )
    initial_state = RigidObjectCfg.InitialStateCfg()

    for i in range(num_steps):
        key = "step_" + str(i)
        current_prim_path = "/World/envs/env_.*/" + key 
        collection[key] = RigidObjectCfg(
            prim_path=current_prim_path,
            spawn=spawn_cfg,
            init_state=initial_state,
            collision_group=0,
            debug_vis=False,
    )

        
    return RigidObjectCollectionCfg(rigid_objects=collection)

```
in `_setup_scene(self)`, I just do
```
self.steps = RigidObjectCollection(self.cfg.steps)
```

However, it seems for each stepping stone, one new physics material is created. I would like to ask what is the correct way of creating one physics material and reusing it for every rigid object in the rigid object collection.

Also, I wonder if I'm using `RigidObjectCollection` in the correct way as every rigid object in it is technically the same but just with different prim paths. Any suggestions on this are appreciated!

RandomOakForest, 2025-04-11:
Thanks for posting this. It seems including `physics_material` may be a causing this. Could you remove it from your parameters for `CuboidCfg` and try again? Thanks.

xindonglin99, 2025-04-11:
Hi @RandomOakForest ,

Yes, removing the attribute `physics_material` can solve the issue. However, what if I want to apply a specific amount of friction to this cube? How can I do that?

Thanks for the reply!

xindonglin99, 2025-05-14:
@RandomOakForest Sorry to ping you again. Is there any update on this? Any suggestions are appreciated.

RandomOakForest, 2025-05-14:
Thanks for following up. So far I have confirmed that there is such a limit, however it should be really hard to reach it. If you use a small number of environments, do you still see the issue? Alternatively, it seems the _for_-loop in your code may be redone following this tutorial on [spawning multiple assets](https://isaac-sim.github.io/IsaacLab/main/source/how-to/multi_asset_spawning.html). I'll post here if I have new findings.

RandomOakForest, 2025-05-14:
I'm moving this issue to our _Discussions_ section for follow up.