isaac-sim/IsaacLab #2289: [Bug Report] Policy Learning Fails When Registering Object in scene.rigid_objects

pernst1001, 2025-04-10:
### Describe the Bug

I created a direct workflow RL environment to train a cube to reach a goal position. Initially, this worked correctly, and the learned policy performed as expected. However, when I attempted to add domain randomization, I needed to make two changes:
1. Add the domain randomization class
2. Add the cube to the scene in `_setup_scene` using: `self.scene.rigid_objects["cube"] = self.cube`

After adding just the second line (before implementing domain randomization), the policy stopped working correctly, despite no other code changes.

**Demonstration Videos:**
1. Working policy (with line commented out):  

https://github.com/user-attachments/assets/5f15553e-f780-41da-bd49-95dd82d5a6f6

   (Note: The policy isn't perfect - it stops moving when the cube is close to the goal - but this is unrelated to the current issue)

2. Non-working policy (with line added):  

https://github.com/user-attachments/assets/0281f359-3a47-4321-9c71-e6f1f515d9a1

#### Expected Behavior
The policy should work identically whether or not the cube is explicitly added to `scene.rigid_objects`.
#### Actual Behavior
The policy fails to learn correctly when the cube is added to `scene.rigid_objects`.

### Steps to Reproduce
1. Copy the [cube folder](https://github.com/pernst1001/small_deformable_error/tree/main/cube) into `IsaacLab/source/isaaclab_tasks/isaaclab_tasks/direct`
2. Launch training:
   ```bash
   ./isaaclab.sh -p scripts/reinforcement_learning/skrl/train.py --task Isaac-cube-v0 --headless --num_envs 64
3. Observe the policy:
   ```bash
   ./isaaclab.sh -p scripts/reinforcement_learning/skrl/play.py --task Isaac-cube-v0 --num_envs 64
4. In cube.env.py, comment out line 62:
   ```python
   self.scene.rigid_objects["cube"] = self.cube
5. Relaunch training and observe the policy again using the same commands
   ```bash
   ./isaaclab.sh -p scripts/reinforcement_learning/skrl/play.py --task Isaac-cube-v0 --num_envs 64

### System Info

- Commit: 6b794ac
- Isaac Sim Version: Isaac Sim 4.5 & Isaac Lab 2.0
- OS: Ubuntu 20.04
- GPU: RTX A4000
- CUDA: 12.2
- GPU Driver: 535.183.01

### Additional context

This issue occurs before implementing domain randomization - the mere addition of the cube to `scene.rigid_objects` affects policy learning.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

- [ ] Identify why adding the cube to `scene.rigid_objects` affects policy learning
- [ ] Provide a solution that allows both proper policy learning and domain randomization implementation

RandomOakForest, 2025-04-10:
Thanks for posting this. Interesting. The team will review.

pernst1001, 2025-04-10:
I agree that it looks like that in the video, but I think that this is just the learnt policy. When I use the random agent from `scripts/environments/random_agent.py` the agents do not move as one, as can be seen in this video

https://github.com/user-attachments/assets/d09fabe1-290d-452b-a123-7b2fed4d5543

AntoineRichard, 2026-08-27:
Thanks @pernst1001 — a training regression that only appears when an object is registered in `scene.rigid_objects` is a genuinely odd one to report, and we appreciate the videos.

We're closing this as not actionable in its current form: the repro needs a full skrl run in an external repo, and it predates the 3.0 backend split and the `InteractiveScene`/`DirectRLEnv` rework. One pointer if you revisit it — registration isn't inert: `InteractiveScene` calls `write_data_to_sim()` and `update()` on the asset every step, and `reset()` on every env reset, which can double-apply resets or clear externally applied forces. Happy to reopen with a minimal non-RL script showing divergent object states on 3.x.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard