isaac-sim/IsaacLab #2297: [Bug Report] Setting the default ground plane color to white appears bluish

dibyabiva, 2025-04-11:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

The default appearance of the default ground plane is black, with white grid.
But setting the color to white makes it appear bluish.

### Steps to reproduce

```
ground = AssetBaseCfg(prim_path = "/World/defaultGroundPlane", spawn = sim_utils.GroundPlaneCfg(color = (1.0, 1.0, 1.0)))
```

Also, how can we set the grid color through code?
I need to set black grid lines on a white ground plane.

### System Info

Describe the characteristic of your environment:

- Commit: f1a4975eb7bae8509082a8ff02fd775810a73531
- Isaac Sim Version: Didn't install separately. It's whatever came with the IsaacLab commit.
- OS: Ubuntu 22.04.5
- GPU: RTX 4070 Ti Super
- CUDA: 12.4.127
- GPU Driver: 570.124.06

### Additional context

Add any other context about the problem here.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

- [ ] Instruction on how to set the ground plane color to white through code, without it looking bluish.
- [ ] Instruction on how to set the grid color to black through code.

RandomOakForest, 2025-04-11:
Thank you for posting this. Could you add a snapshot of your GUI with this ground plane color you are observing? Thanks.

dibyabiva, 2025-04-11:
Here's what it looks like when I set the color to white (1.0, 1.0, 1.0)

[image]

dibyabiva, 2025-04-15:
Thanks @Bardreamaster , but I was looking for a way to change it through code.

I just created a thin cuboid USD in Blender and imported it for now.

I'll still be looking out for ways to change the ground and grid colors through code.

Bardreamaster, 2025-04-16:
@dibyabiva Yes, you can, change `albedo_add` after this function:

https://github.com/isaac-sim/IsaacLab/blob/7de6d6fef9424c95fc68dc767af67ffbe0da6832/source/isaaclab/isaaclab/sim/spawners/from_files/from_files.py#L165

by adding:

```python
        prop_path = f"{prim_path}/Looks/theGrid/Shader.inputs:albedo_add"
        omni.kit.commands.execute(
            "ChangePropertyCommand",
            prop_path=Sdf.Path(prop_path),
            value=1.0,
            prev=None,
            type_to_create_if_not_exist=Sdf.ValueTypeNames.Float,
        )
```

Then, use the way you change color before, the color view should be the same as you expect.

dibyabiva, 2025-04-16:
@Bardreamaster 

The snippet you provided doesn't work in an [Interactive Scene](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/02_scene/create_scene.html). I end up getting this error:
```
2025-04-16 03:13:16 [11,876ms] [Error] [omni.kit.commands.command] Failed to execute a command: ChangePropertyCommand.
  File "/home/dibyabiva/Desktop/work/isaac-lab/custom_scripts/create_data_from_simulation.py", line 449, in <module>
    main()
  File "/home/dibyabiva/Desktop/work/isaac-lab/custom_scripts/create_data_from_simulation.py", line 427, in main
    omni.kit.commands.execute(
  File "/home/dibyabiva/Desktop/work/isaac-lab/env/lib/python3.10/site-packages/isaacsim/extscache/omni.kit.commands-1.4.9+d02c707b.lx64.r.cp310/omni/kit/commands/command.py", line 463, in execute
    result = omni.kit.undo.execute(command, name, kwargs)
[...skipped...]
  File "/home/dibyabiva/Desktop/work/isaac-lab/env/lib/python3.10/site-packages/isaacsim/extscache/omni.kit.commands-1.4.9+d02c707b.lx64.r.cp310/omni/kit/undo/undo.py", line 78, in execute
    result = _execute(command, name, level, history_key)
  File "/home/dibyabiva/Desktop/work/isaac-lab/env/lib/python3.10/site-packages/isaacsim/extscache/omni.kit.commands-1.4.9+d02c707b.lx64.r.cp310/omni/kit/undo/undo.py", line 459, in _execute
    raise error
  File "/home/dibyabiva/Desktop/work/isaac-lab/env/lib/python3.10/site-packages/isaacsim/extscache/omni.kit.commands-1.4.9+d02c707b.lx64.r.cp310/omni/kit/undo/undo.py", line 420, in _execute
    result = command.do()
  File "/home/dibyabiva/Desktop/work/isaac-lab/env/lib/python3.10/site-packages/isaacsim/extscache/omni.usd-1.12.4+d02c707b.lx64.r.cp310/omni/usd/commands/usd_commands.py", line 4088, in do
    current_layer = self._get_target_layer()
  File "/home/dibyabiva/Desktop/work/isaac-lab/env/lib/python3.10/site-packages/isaacsim/extscache/omni.usd-1.12.4+d02c707b.lx64.r.cp310/omni/usd/commands/usd_commands.py", line 4043, in _get_target_layer
    if prop.HasCustomDataKey("nonpersistant") and prop.GetCustomDataByKey("nonpersistant"):

 <class 'RuntimeError'> Accessed <pxr.Usd.Property object at 0x77a445b72b40>
```

Bardreamaster, 2025-04-16:
@dibyabiva  Could you try it on the latest IsaacLab 7de6d6fef9424c95fc68dc767af67ffbe0da6832 ? I've test by running the [tutorial](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/02_scene/create_scene.html) with only ground plane color changes. Run with  `./isaaclab.sh -p scripts/tutorials/02_scene/create_scene.py`

[image]

And this file ([IsaacLab/source/isaaclab/isaaclab/sim/spawners/from_files/from_files.py](https://github.com/isaac-sim/IsaacLab/blob/7de6d6fef9424c95fc68dc767af67ffbe0da6832/source/isaaclab/isaaclab/sim/spawners/from_files/from_files.py#L165)) should be changed into like this:

```python
    # ...Existing codes

    if cfg.color is not None:
        prop_path = f"{prim_path}/Looks/theGrid/Shader.inputs:diffuse_tint"
        # change the color
        omni.kit.commands.execute(
            "ChangePropertyCommand",
            prop_path=Sdf.Path(prop_path),
            value=Gf.Vec3f(*cfg.color),
            prev=None,
            type_to_create_if_not_exist=Sdf.ValueTypeNames.Color3f,
        )
        prop_path = f"{prim_path}/Looks/theGrid/Shader.inputs:albedo_add"
        omni.kit.commands.execute(
            "ChangePropertyCommand",
            prop_path=Sdf.Path(prop_path),
            value=1.0,
            prev=None,
            type_to_create_if_not_exist=Sdf.ValueTypeNames.Float,
        )
    # Remove the light from the ground plane
    # It isn't bright enough and messes up with the user's lighting settings
    omni.kit.commands.execute("ToggleVisibilitySelectedPrims", selected_paths=[f"{prim_path}/SphereLight"])

    # ...Existing codes
```

dibyabiva, 2025-04-16:
@Bardreamaster I had misunderstood it the first time. I was editing my script, instead of the IsaacLab source code [IsaacLab/source/isaaclab/isaaclab/sim/spawners/from_files/from_files.py](https://github.com/isaac-sim/IsaacLab/blob/7de6d6fef9424c95fc68dc767af67ffbe0da6832/source/isaaclab/isaaclab/sim/spawners/from_files/from_files.py#L165)

I was looking for a way to change it without changing the source code.

However, I followed your steps, and it works. 

I guess there's also a similar way to change the color of the grid-lines.

Thanks for this info. This will surely come in handy some time.

Bardreamaster, 2025-04-16:
@dibyabiva I understand. The error you meet before might because you try to edit Property in `SceneCfg`. The context did not initialize at that time.

Use the tutorial as an example, you could place this snippet at the beginning of the `def run_simulator(sim: sim_utils.SimulationContext, scene: InteractiveScene):` like this:

```python
def run_simulator(sim: sim_utils.SimulationContext, scene: InteractiveScene):
    """Runs the simulation loop."""
    prop_path = f"/World/defaultGroundPlane/Looks/theGrid/Shader.inputs:albedo_add"
    omni.kit.commands.execute(
        "ChangePropertyCommand",
        prop_path=Sdf.Path(prop_path),
        value=1.0,
        prev=None,
        type_to_create_if_not_exist=Sdf.ValueTypeNames.Float,
    )

# ...exist codes
```

Then you don't have to edit isaaclab source code.

Do not forget necessary imports:

```python
import omni.kit.commands
from pxr import Sdf
```

dibyabiva, 2025-04-16:
@Bardreamaster 
Thank you so much, this worked.

Can you also tell me how to change the grid color?
I need black grid lines on white ground.

Bardreamaster, 2025-04-16:
@dibyabiva I am not sure if there's one way to change grid color without change the plane texture. I think you have to change a plane with different texture like [these exist assets](https://docs.isaacsim.omniverse.nvidia.com/latest/assets/usd_assets_environments.html#id1).

And actually, we do **not** really change the color of the **ground plane**. The plane should be like [wireframe with blue texture](http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.5/Isaac/Environments/Grid/Materials/Textures/Wireframe_blue.png).

The `albedo_add` just makes the plane reflects all lights so the camera in simulator recognize the plane as the color we set. While the grid is white, so any other added color into white light still looks like white.

So I think preparing a plane asset meet your needs is a good way for you.

dibyabiva, 2025-04-16:
@Bardreamaster Thank you for the in-depth clarification.
Yes, that's what I ended up doing ultimately. Just wanted to make sure that I wasn't missing something obvious.

Thank you once again for the prompt responses.

closed 2025-04-16 by dibyabiva