isaac-sim/IsaacLab #2301: performance issue with blackwell GPU

RX-02333, 2025-04-12:
Hello，I use a RTX5090d to run isaac lab benchmark,and I found something performance issue with it.
my env is  windows 11 and cuda12.8,pytorch nightly version,driver version 572.83.
I run python scripts/benchmarks/benchmark_rlgames.py --task=Isaac-Cartpole-v0 --headless and compare it with 4090,just 1/2 performace of 4090:

[image]
what should I do for fix this issue?

RX-02333, 2025-04-14:
> Thank you for posting this. _On Windows with CUDA 12, the GPU driver version 552.86 is required._

Thank your reply！I indeed use driver version 552.86：

[image]

RandomOakForest, 2025-06-02:
Following up on this, let us know if you still need further help, and if so, please describe the specific issues you still observe. I'll move this post to our _Discussions_ for follow up. Thank you for your interest in Isaac Lab.