isaac-sim/IsaacLab #2304: [Question] Deformable object isn't lifted

songwookim, 2025-04-12:
### Question

https://github.com/user-attachments/assets/d4e28fd1-800b-4f70-907f-edd3bc03f007

```python  
# .../lift/config/franka/joint_pos_env.cfg.py
        self.scene.object = DeformableObjectCfg(
            prim_path="{ENV_REGEX_NS}/Object",
            init_state=DeformableObjectCfg.InitialStateCfg(pos=[0.5, 0, 0.055], rot=[1, 0, 0, 0]), # type: ignore            
            spawn=sim_utils.MeshCuboidCfg(
                size=(0.04, 0.04, 0.04),
                deformable_props=DeformableBodyPropertiesCfg(rest_offset=0.0,contact_offset=0.001),
                physics_material=sim_utils.DeformableBodyMaterialCfg(poissons_ratio=0.4, youngs_modulus=1e5),
                visual_material=sim_utils.PreviewSurfaceCfg(diffuse_color=(0.5, 0.1, 0.0)),
                mass_props=sim_utils.MassPropertiesCfg(
                    mass=0.1,
                ),
            )
```
Hi guys, in the Lift environment, we were replacing the existing cube with a deformable cube. This is the screen when running 'teleop_se3_agent.py', but the Franka robot fails to lift the cube. 

The code above shows the parameters for DeformableObjectCfg in self.scene.object.
Below is the code we tried, but it didn't work.

Could I ask about the possible cause or any hints to solve this issue?

```python
# .../teleoperation/teleop_se3_agent.py
    object_path = "/World/envs/env_0/Object"
    obstacle_prim = stage_utils.get_current_stage().GetPrimAtPath(object_path)
    mass_body = UsdPhysics.MassAPI.Apply(obstacle_prim)
    mass_body.CreateMassAttr().Set(1.0)
    deformableUtils.add_physx_deformable_body(
        stage_utils.get_current_stage(),
        object_path,
        collision_simplification=True,
        simulation_hexahedral_resolution=8,
        self_collision=False,
        solver_position_iteration_count=20
    )
```
and These are development settings
- Isaac lab version : 2.3.0
- Isaac sim version : 4.5
- ubuntu / gpu : 22.04 / rtx4090

songwookim, 2025-04-12:
i got solution by adding these
```python
vertex_velocity_damping = 0.1  # add motion equilibrium
solver_position_iteration_count = 50  # add collision accuracy
```

RandomOakForest, 2025-04-14:
Thanks for posting this. It seems friction is missing, or the object is too flexible. I'll move this post to our discussions for the team to follow up.