isaac-sim/IsaacLab #2307: Flywheel decelerates in frictionless environment with only positive DCMotor torque

VoytekK, 2025-04-14:
### Describe the bug

In a frictionless simulation environment using Isaac Sim, a flywheel driven by a DCMotor actuator **decelerates** unexpectedly when the torque decreases but remains positive. This issue was identified when tuning a custom actuator, but I recreated it by editing a DCMotor actuator to only return positive torque outputs. All friction and damping settings are set to zero. Despite this, the flywheel slows down when torque is reduced but still positive, which should only allow acceleration or constant velocity, not deceleration. Once torque reaches zero, the velocity stabilizes, indicating no friction is present. This behavior suggests an unintended source of damping or friction in the simulation.

### Steps to reproduce
Create a flywheel, I defined mine using a MuJoCo xml file, with the key items being:
```
<body name="flywheel" pos="-0.0125 0 0" material="flywheel_material">
    <joint name="motor_joint" axis="1 0 0" type="hinge"/>
    <geom name="flywheel_geom" size="0.06 0.012"  density="3508.482" type="cylinder"  euler="0 90 0"/>
</body>
```
After importing the flywheel into Isaac Sim, ensure all friction related items like Angular Damping are set to zero and save as a usd file.  

Create the configuration for the flywheel and connect a DCMotor to drive it:
```
FLYWHEEL_CFG = ArticulationCfg(
    soft_joint_pos_limit_factor=0.95,
    spawn=sim_utils.UsdFileCfg(
        usd_path=__file__.replace("base.py", "flywheel.usd"),
        activate_contact_sensors=True,
        rigid_props=sim_utils.RigidBodyPropertiesCfg(
            disable_gravity=False,
            retain_accelerations=False,
            linear_damping=0.0,
            angular_damping=0.0,
            max_linear_velocity=100.0,
            max_angular_velocity=100,
            max_depenetration_velocity=1.0,
        ),
        articulation_props=sim_utils.ArticulationRootPropertiesCfg(
            enabled_self_collisions=False,
            solver_position_iteration_count=100,
            solver_velocity_iteration_count=100,
        ),
    ),
    init_state=ArticulationCfg.InitialStateCfg(
        pos=(0.0, 0.0, 0.0),
        joint_pos={
            "motor_joint": 0.0,
        },
        joint_vel={".*": 0.0},
    ),
    actuators={
        "flywheel": DCMotorCfg(
            joint_names_expr=["motor_joint"],
            effort_limit=5,  
            saturation_effort=5,  
            velocity_limit=20, 
            stiffness=15, 
            damping=0.1, 
            friction=0.0, 
            velocity_limit_sim=500,
        ),
    },
)
```

Tweak the DCMotor actuator to never be able to return a negative torque. This ensures that flywheel velocity cannot be reduced with any forces applied by DCMotor. In `actuator_pd.py`, update the `compute` function in the`IdealPDActuator` class.
```
def compute( ....)
    ....
    zero = torch.tensor([[0]], device="cuda")
    self.applied_effort = torch.clip(self.computed_effort, min=zero, max=self.effort_limit)

    control_action.joint_efforts = self.applied_effort
    control_action.joint_positions = None
    control_action.joint_velocities = None

    return control_action
```

### Results
With the changes made, run a simulation that applies a positive position request to the joint. Observe the velocity of the flywheel. As the actual position approaches the desired position, the torque is reduced and decays to zero. During the torque decay, velocity decreases following the exact curve of the torque decay. See the image blow. 

[image]

#### Image Explantion:

**Subplot 3 - Position** 
The DCMotor is requested to drive to a position just under 20k (this is 4.5 revolutions with an encoder count of 4096 counts per revolution as I was modeling hardware). (Red Trace)
The motor position increases to track to the desired position (Blue Trace)

**Subplot 1 - Torque**
The DCMotor actuator applies the maximum torque is is allowed (5Nm) to the flywheel for about 100ms. At this point the torque starts to decrease as we approach the desired position. 

**Subplot 2 - Velocity**
At time zero, velocity is zero, but difficult to see in this simulation. There are about 3 data points showing the velocity increase from zero to it's maximum. With the applied motor torque and flywheel inertial this is expected. 
The velocity reaches a maximum, which appears correct as there is a `velocity_limit` set. 
What is completely wrong is that at about 100ms Velocity DECREASES. This is a frictionless world! What is most strange is that the velocity decrease curve is identical to the torque decrease curve. The velocity starts to make sense again after about 170ms when the applied torque is set to zero. At this point the velocity stops decreasing, show a frictionless world. 

 

### System Info

Describe the characteristic of your environment:
- Commit: 09590912792d4421c84f053c5c13c31391bb5c30
- Isaac Sim Version:
  - App Version: 4.5.0-rc.36
  - Kit SDK Version: 106.5.O
  - Client Library Version: 2.55.0
- OS: Windows 11 Pro 23H2
- GPU: NVIDIA GeForce RTX 4060 Laptop GPU" (8 GiB, sm_89, mempool enabled)
- CUDA: Toolkit 11.8, Driver 12.5
- GPU Driver: NVIDIA-SMI 556.35

### Checklist

- [X] I have checked that there is no similar issue in the repo (**required**)
- [X] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Newton's first law of motion is respected. 
- [ ] A positive torque should always accelerate a mass in a frictionless environment, not slow it down.
- [ ] Isaac Sim/Lab do not apply an invisible torque that slows an object.

RandomOakForest, 2025-04-15:
Thank you for posting this and adding this level of detail. This is a very interesting catch. We will review it.

VoytekK, 2025-04-27:
I found one of the causes of this. Almost everything in Isaac Sim is in radians, except for:
`max_angular_velocity=100`
This value is in degrees, which is much too slow for this test. This is the velocity that the flywheel drops down to when it is coasting. 

What is still an issue is that the simulation allows the angular velocity to exceed the 100 degrees per second set by `max_angular_velocity` while torque is applied. Then when torque is no longer applied the value of `max_angular_velocity` is respected. 

Setting the wrong value for my simulation is my bug, but the velocity exceeding the maximum allowed, and then following the torque curve seems like a bug to me.

RandomOakForest, 2025-04-29:
Thank you for following up. We are reviewing why the simulation would allow the angular velocity to go over `max_angular_velocity` as implemented in Isaac Lab.

RandomOakForest, 2025-06-02:
Following up on this, I will close this issue for now, but please feel free to reopen it may you need further help. Thank you.

closed 2025-06-02 by RandomOakForest