isaac-sim/IsaacLab #2320: Unable to start AR in isaac lab by using handtracking device(oculus quest pro)

vaacuum-song, 2025-04-16:
Question:
   Hi:
   When executing the command:
   ./isaaclab.bat -p scripts/environments/teleoperation/teleop_se3_agent.py --task Isaac-Lift-Cube-Franka-IK-Rel-v0 - 
   -num_envs 1 --teleop_device handtrackingto control the Franka arm via Oculus Quest Pro in AR mode (the START 
   VR mode works normally), clicking the ‌START AR‌ button in the Isaac Sim window triggers a console error. The 
   error log suggests compatibility or configuration issues with the AR module, and the log is attached below.

Steps to Reproduce
   run ./isaaclab.bat -p scripts/environments/teleoperation/teleop_se3_agent.py --task Isaac-Lift-Cube-Franka-IK-Rel- 
   v0 --num_envs 1 --teleop_device handtracking

   click START AR in isaacsim window

   throw an error

Error Messages
   2025-04-15 02:56:24 [23,972ms] [Error] [omni.kit.xr.system.openxr.plugin] Failed to find system with HMD form 
   factor/ View configuration

   2025-04-15 02:56:24 [24,098ms] [Error] [omni.kit.xr.ui.window.viewport.viewport_layer.viewport_layer] Failed to 
   activate OpenXR: Failed to find system with HMD form factor/ View configuration

[kit_20250414_000001.log](https://github.com/user-attachments/files/19768748/kit_20250414_000001.log)

RandomOakForest, 2025-04-17:
Thanks for posting this. The team is constantly working to add support for AR modes. Unfortunately, we currently do not support the Oculus Quest Pro (Meta OpenXR).

vaacuum-song, 2025-04-18:
> Thanks for posting this. The team is constantly working to add support for AR modes. Unfortunately, we currently do not support the Oculus Quest Pro (Meta OpenXR).

ok, got it

closed 2025-04-18 by vaacuum-song