isaac-sim/IsaacLab #2323: [Bug Report] Unable to allow collisions between robot and trimesh terrain

5hadytru, 2025-04-16:
I've followed the instructions from issue #2187 , and at this point it appears that the terrain is properly added to the simulation (as evidenced by me seeing the trimesh when I render); however, my robot spawns in and immediately goes straight through the terrain (the robot spawns above it and falls through, which I saw when recording a video). Here is how I import the terrain (in the setup_scene method:

`
physics_material_cfg = sim_utils.RigidBodyMaterialCfg(
            static_friction=1.0,
            dynamic_friction=1.0,
            restitution=0.0,
            friction_combine_mode="multiply",
            restitution_combine_mode="multiply"
        )

        visual_material_cfg = sim_utils.PreviewSurfaceCfg(
            diffuse_color=(0.0, 0.0, 1.0)
        )
        
        self._get_initial_env_origins()

        terrain_importer = TrimeshTerrainImporter(
            vertices=self.terrain.vertices,
            triangles=self.terrain.triangles,
            translation=(-self.terrain.cfg.border_size, -self.terrain.cfg.border_size, 0.0),
            initial_env_origins=self.initial_env_origins, 
            physics_material_cfg=physics_material_cfg, 
            visual_material_cfg=visual_material_cfg,
            device=self.cfg.sim.device
        )
        
        self.scene._terrain = terrain_importer
`

where the TrimeshTerrainImporter is like this:

`
class TrimeshTerrainImporter:
    def __init__(
        self, 
        vertices:np.ndarray,
        triangles:np.ndarray,
        translation,
        initial_env_origins:torch.Tensor, 
        physics_material_cfg, 
        visual_material_cfg,
        device
    ):
        self.vertices = vertices
        self.triangles = triangles
        self.translation = translation
        self.env_origins = initial_env_origins.clone()
        self.visual_material = visual_material_cfg
        self.physics_material = physics_material_cfg
        self.device = device

        self.prim_path = "/World/Terrain"

        self.import_mesh()

    def import_mesh(self):
        """
        Create a trimesh.Trimesh object -> store as USD prim
        """
        mesh = trimesh.Trimesh(self.vertices, self.triangles)
        create_prim_from_mesh(
            self.prim_path, mesh, visual_material=self.visual_material, physics_material=self.physics_material, translation=self.translation
        )
`

and I'm attempting to allow collisions with the terrain like this:

`
self.scene.clone_environments(copy_from_source=False)
self.scene.filter_collisions(global_prim_paths=[self.scene._terrain.prim_path + "/mesh"])
`

One interesting occurance is that, despite the terrain showing up when rendering, I am getting the following Warning (bottom Warning):

`
[INFO]: Scene manager:  <class InteractiveScene>
	Number of environments: 6144
	Environment spacing   : 5.0
	Source prim name      : /World/envs/env_0
	Global prim paths     : ['/World/Terrain_cuda_0/mesh']
	Replicate physics     : True
[INFO]: Starting the simulation. This may take a few seconds. Please wait...
2025-04-16 20:46:02 [116,896ms] [Warning] [omni.physx.cooking.plugin] m_processResult == OperationResult::SUCCESS && "result parameter" is false.
2025-04-16 20:46:02 [116,897ms] [Warning] [omni.physx.cooking.plugin] UjitsoMeshCookingContext: cooking failure
2025-04-16 20:46:02 [116,897ms] [Warning] [omni.physx.plugin] Unable to create triangle mesh for:/World/Terrain_cuda_0/mesh
`

Just in case, here is my robot cfg:

`
GO1_ACTUATOR_CFG = ActuatorNetMLPCfg(
    joint_names_expr=[".*_hip_joint", ".*_thigh_joint", ".*_calf_joint"],
    network_file=f"{ISAACLAB_NUCLEUS_DIR}/ActuatorNets/Unitree/unitree_go1.pt",
    pos_scale=-1.0,
    vel_scale=1.0,
    torque_scale=1.0,
    input_order="pos_vel",
    input_idx=[0, 1, 2],
    effort_limit=23.7,  # taken from spec sheet
    velocity_limit=30.0,  # taken from spec sheet
    saturation_effort=23.7,  # same as effort limit
)

UNITREE_GO1_CFG = ArticulationCfg(
    prim_path="/World/envs/env_.*/Robot",
    spawn=sim_utils.UsdFileCfg(
        usd_path=f"{ISAACLAB_NUCLEUS_DIR}/Robots/Unitree/Go1/go1.usd",
        activate_contact_sensors=True,
        rigid_props=sim_utils.RigidBodyPropertiesCfg(
            rigid_body_enabled=True,
            disable_gravity=False,
            max_linear_velocity=1000.0,
            max_angular_velocity=1000.0,
        ),
        articulation_props=sim_utils.ArticulationRootPropertiesCfg(
            enabled_self_collisions=True, solver_position_iteration_count=4, solver_velocity_iteration_count=0
        ),
    ),
    init_state=ArticulationCfg.InitialStateCfg(
        pos=(0.0, 0.0, 0.42),
        joint_pos={ # = target angles [rad] when action = 0.0
            'FL_hip_joint': 0.1,   # [rad]
            'RL_hip_joint': 0.1,   # [rad]
            'FR_hip_joint': -0.1 ,  # [rad]
            'RR_hip_joint': -0.1,   # [rad]

            'FL_thigh_joint': 0.8,     # [rad]
            'RL_thigh_joint': 1.,   # [rad]
            'FR_thigh_joint': 0.8,     # [rad]
            'RR_thigh_joint': 1.,   # [rad]

            'FL_calf_joint': -1.5,   # [rad]
            'RL_calf_joint': -1.5,    # [rad]
            'FR_calf_joint': -1.5,  # [rad]
            'RR_calf_joint': -1.5,    # [rad]
        },
        joint_vel={".*": 0.0},
    ),
    soft_joint_pos_limit_factor=0.9,
    actuators={
        "base_legs": GO1_ACTUATOR_CFG,
    },
)
`

So to recap:
1. The terrain shows up (proper obstacles, visual material, and rough terrain show up, for example)
2. The robot falls straight through the terrain immediately; no collisions enabled it appears
3. I am getting the Warning "[omni.physx.plugin] Unable to create triangle mesh for:/World/Terrain_cuda_0/mesh"

RandomOakForest, 2025-04-17:
Thanks for posting this. It is possible that you are running into a memory issue. Could you try less environments or a terrain that is somewhat simpler?

5hadytru, 2025-04-21:
Ok looks like decreasing the resolution of the trimesh (number of triangles/vertices) solved the problem. Note, however, that the "Isaac Lab way" of adding the trimesh to the scene appears to take *way* more memory than Isaac Gym; in Gym, I could use the full-resolution trimesh as my terrain on a <100GB RAM machine. I am now on a 440GB RAM machine (Azure A10) and had this issue. Also note that decreasing the number of envs from 6144 to 2 didn't help, assumedly since the terrain trimesh is a global prim. Thanks for the help!

closed 2025-04-21 by 5hadytru