isaac-sim/IsaacLab #2338: [Question] How to properly use `RigidBodyView`'s `set_disable_simulations` function with Isaac Lab

zoctipus, 2025-04-19:
### Question

I am interested in using IsaacLab manager based env to simulate large number of `sdf` collider object in each environment (about 25 per env), and the simulation gets very slow as number of `sdf` colliders are increasing in each env. Depending on task assigned to each env, some assets can be more or less relevant, exploring that direction, I found out `RigidBodyView.set_disable_simulations`  at https://docs.omniverse.nvidia.com/kit/docs/omni_physics/latest/extensions/runtime/source/omni.physics.tensors/docs/api/python.html#omni.physics.tensors.impl.api.RigidBodyView.set_disable_simulations

however, I don't think it works with default isaac lab set up.

below is my snippet in disabling the assets
```
def disable_assets(
    env: ManagerBasedRLEnv,
    env_ids: torch.Tensor,
    asset_cfg: SceneEntityCfg,
    asset_reset_key_points_cfg: TaskDataCfg,
):
    assets: RigidObjectCollection = env.scene[asset_cfg.name]
    disable_simulation_mask = assets.reshape_view_to_data(assets.root_physx_view.get_disable_simulations().to(env.device))
    disable_simulation_mask[:] = 1
    disable_simulation_mask[env_ids[:, None], assets_ids[env_ids]] = 0

    view_data = assets.reshape_data_to_view(disable_simulation_mask)
    view_indices = assets._env_obj_ids_to_view_ids(env_ids, object_ids=slice(None))
    assets.root_physx_view.set_disable_simulations(view_data.to('cpu'), view_indices.to('cpu'))
```

while the function call will work, the simulation quickly enters Error shown below

[image]

I do realize that the solution to this can be very complex(Maybe illegal to access rigidbody data pertaining these assets once they are disabled.), I wonder if you have any other suggestions in sleeping/disabling parts of the asset and improve optimization? 

Thanks!

RandomOakForest, 2025-04-20:
Thanks for posting this. Currently, there isn’t a way to change the asset properties at runtime right now, it’ll have to be done before simulation starts. This may be causing the error you are getting. Are you sure your code is not trying to perform a change once simulation has started?

zoctipus, 2025-04-21:
the code snippet is perform at "reset" mode not "pre-startup" mode, so I think you are right that I should not call set_disable_simulations during simulation.

It does feel like being able to set some object to sleep will be a good feature during runtime can be beneficial to many application, though, it seems to require the Isaac Sim side to support this feature, and it is hard to predict how difficult it is. Feel free to change this to enhancement or close it upon your judgement!