isaac-sim/IsaacLab #2342: [Bug Report] Type check with class based manager term but

miguelalonsojr, 2025-04-20:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

In the event manager, class based event terms are not registered properly due to a missing `type()` call. On [line 387 ](https://github.com/isaac-sim/IsaacLab/blob/7de6d6fef9424c95fc68dc767af67ffbe0da6832/source/isaaclab/isaaclab/managers/event_manager.py#L387) of `event_manager.py`, the check should be

```            
            if isinstance(term_cfg.func, type(ManagerTermBase)):
```

otherwise, `isinstance` always returns false (since `ManagerTermBase` is an abstract base class, the `isintance` method will check that `term_cfg.func` is derived from `ABC`, and not `ManagerTermBase`). Adding `type(ManagerTermBase)` resolves the issue.

### Steps to reproduce

1. Define a class based event manager term.
2. Add it as an event term in when creating a simulation environment.
3. Set a debug breakpoint at line 387 and check the conditional statement.

### System Info

Describe the characteristic of your environment:

- Commit: 7de6d6fef9424c95fc68dc767af67ffbe0da6832
- Isaac Sim Version: 2.0.2 (main branch)
- OS: Ubuntu 22.04
- GPU: RTX 4090
- CUDA: 12.1
- GPU Driver:  550.120 

### Additional context

Add any other context about the problem here.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Class based event terms are registered correctly and the __call__ and reset() functions are called properly while stepping an environment.

RandomOakForest, 2025-04-21:
Thank you for posting this. Interesting catch. The team will review.

zoctipus, 2025-04-21:
I also faced this bug currently, having create bug or pull request yet, I hot-fixed by changing that line to 
```
if inspect.isclass(term_cfg.func):
```

miguelalonsojr, 2025-04-21:
> I also faced this bug currently, have not create issue yet, I hot-fixed by changing that line to
> 
> ```
> if inspect.isclass(term_cfg.func):
> ```

Yes, this will work as well. For now, I've patched it locally. But it would be great to have this fixed. @RandomOakForest Can I contribute a PR? It's a one line fix. I'd be happy to submit it.

kellyguo11, 2025-04-21:
We'll be happy to take a PR for the fix. Thanks for spotting the bug!

miguelalonsojr, 2025-04-23:
Great. Will submit a PR today. Thanks!

closed 2025-04-25 by kellyguo11