isaac-sim/IsaacLab #2347: [Question] Adding friction properties to terrain imported from .obj file

saichand44, 2025-04-21:
Hi,

For my project, I am using a non-planar surface (generated using python script + blender). I have successfully converted the .obj file to .usd using `MeshConverterCfg`.

```
non_planar_terrain_path = f"{ROBORACER_ASSETS_DATA_DIR}/terrains/bump_track.obj"
usd_filename = non_planar_terrain_path.split("/")[-1].split(".")[0]
dest_usd_path = f"{ROBORACER_ASSETS_DATA_DIR}/terrains/{usd_filename}/"

NON_PLANAR_TERRAIN_CFG = MeshConverterCfg(
    mass_props=None,
    rigid_props=None,
    collision_props=sim_utils.CollisionPropertiesCfg(collision_enabled=True),
    asset_path=non_planar_terrain_path,
    force_usd_conversion=True,
    usd_dir=dest_usd_path,
    usd_file_name=usd_filename,
    make_instanceable=True,
    collision_approximation="meshSimplification",
    translation=(0.0, 0.0, 0.0),
    rotation=(0.5, 0.5, -0.5, -0.5),
    scale=(1.0, 1.0, 1.0),
)
```

Then, I used it with `TerrainImporterCfg` using the following command
```
    terrain = TerrainImporterCfg(
        prim_path="/World/NonPlanarSurface",
        terrain_type = "usd",
        usd_path=NON_PLANAR_TERRAIN_CFG.usd_dir + NON_PLANAR_TERRAIN_CFG.usd_file_name + ".usd",

        # This parameter is used only when the terrain_type is “generator” or “plane”.
        physics_material=sim_utils.RigidBodyMaterialCfg(
            friction_combine_mode="average",
            restitution_combine_mode="average",
            static_friction=1.0,
            dynamic_friction=1.0,
        ),
        debug_vis=True
    )
```

But from the documentation of `TerrainImporterCfg`, I found that `physics_material` is only applicable when  `terrain_type` is “generator” or “plane”. 

I want to avoid defining the friction properties in Isaac Sim GUI or blender; instead, I want to define them within the Python script I currently use for Isaac Sim. I would appreciate if I can get some direction on this.

RandomOakForest, 2025-04-25:
Thanks for posting this question. The properties would have to be defined in the USD file. We area aware of the advantage of doing this programatically, and the team will review this as a proposal towards a possible enhancement in the future.

maxkra15, 2026-10-02:
Thanks @saichand44 — you can set the imported terrain's friction entirely from Python.
`TerrainImporterCfg.physics_material` still applies only to generated/plane terrain.
For USD terrain, run this after import and before initializing the simulation:

```python
import isaaclab.sim as sim_utils
from isaaclab.sim.spawners.materials import (
    UsdPhysicsRigidBodyMaterialCfg,
    spawn_physics_material,
)

terrain_path = terrain.terrain_prim_paths[0]
material_path = f"{terrain.cfg.prim_path}/physicsMaterial"
sim_utils.make_uninstanceable(terrain_path)
spawn_physics_material(
    material_path,
    UsdPhysicsRigidBodyMaterialCfg(static_friction=1.0, dynamic_friction=1.0),
)
sim_utils.bind_physics_material(terrain_path, material_path)
```

Here `terrain` is your `TerrainImporter` instance. Disabling instanceability
makes its collider descendants editable and can increase stage memory.
I verified this with an imported instanceable mesh on current `develop`: its
collider resolves to the new physics material with both friction coefficients set
to 1.0. The `bind_physics_material()` helper applies it recursively to physics-enabled prims.

Closing as answered. Please reopen if this workflow fails on current Isaac Lab
and Isaac Sim 6.x, with a minimal USD/script.

closed 2026-10-02 by maxkra15