isaac-sim/IsaacLab #2348: [Bug Report] rsl_rl train not working - rnd_state not found in infos['observations']

NoahSTEMStuff, 2025-04-21:
### Describe the bug

I noticed when increasing just the history length of my observations and nothing else, my policies were drastically different. I have modified the g1 locomotion config and usd, and I got a decent policy with a history length of 2, but increasing it to three yielded a much worse walking gait. This didn't make sense to me, so I thought I must have accidentally modified a file. Thus I updated the Isaac lab files to the latest commit, but now I cannot train at all.

### Steps to reproduce

I ran rsl_rl/train.py as follows:
```~/IsaacLab$ ./isaaclab.sh -p scripts/reinforcement_learning/rsl_rl/train.py --task=modified-Velocity-Rough-G1-v1 --headless --video --video_length 100 --video_interval 4000 --enable_cameras```

In the interest of brevity, I removed some of the terminal output.
```
[INFO] Using python from: /home/noah/IsaacLab/_isaac_sim/python.sh                                                                                                       
[INFO][AppLauncher]: Loading experience file: /home/noah/IsaacLab/apps/isaaclab.python.headless.rendering.kit
Loading user config located at: '/home/noah/isaacsim/kit/data/Kit/Isaac-Sim/4.5/user.config.json'
[Info] [carb] Logging to file: /home/noah/isaacsim/kit/logs/Kit/Isaac-Sim/4.5/kit_20250418_142553.log
2025-04-18 21:25:53 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.

[INFO]: Parsing configuration from: isaaclab_tasks.manager_based.locomotion.velocity.config.g1.rough_env_cfg_modified_v1:G1RoughEnvCfg
[INFO]: Parsing configuration from: isaaclab_tasks.manager_based.locomotion.velocity.config.g1.agents.rsl_rl_ppo_cfg:G1RoughPPORunnerCfg
[INFO] Logging experiment in directory: /home/noah/IsaacLab/logs/rsl_rl/g1_rough
Exact experiment name requested from command line: 2025-04-18_14-27-25
Setting seed: 42
[INFO]: Base environment:
	Environment device    : cuda:0
	Environment seed      : 42
	Physics step-size     : 0.005
	Rendering step-size   : 0.02
	Environment step-size : 0.02
[INFO] Generating terrains based on curriculum took : 1.073209 seconds
[INFO]: Time taken for scene creation : 5.940598 seconds
[INFO]: Scene manager:  <class InteractiveScene>
	Number of environments: 4096
	Environment spacing   : 2.5
	Source prim name      : /World/envs/env_0
	Global prim paths     : ['/World/ground']
	Replicate physics     : True
[INFO]: Starting the simulation. This may take a few seconds. Please wait...
[INFO]: Time taken for simulation start : 61.118775 seconds

2025-04-18 21:28:36 [162,344ms] [Warning] [omni.hydra.scene_delegate.plugin] Calling getBypassRenderSkelMeshProcessing for prim /Visuals/Command/velocity_current.proto0_Cone_id1 that has not been populated
ActorCritic.__init__ got unexpected arguments, which will be ignored: ['noise_std_type']
Actor MLP: Sequential(
  (0): Linear(in_features=198, out_features=512, bias=True)
  (1): ELU(alpha=1.0)
  (2): Linear(in_features=512, out_features=256, bias=True)
  (3): ELU(alpha=1.0)
  (4): Linear(in_features=256, out_features=128, bias=True)
  (5): ELU(alpha=1.0)
  (6): Linear(in_features=128, out_features=29, bias=True)
)
Critic MLP: Sequential(
  (0): Linear(in_features=572, out_features=512, bias=True)
  (1): ELU(alpha=1.0)
  (2): Linear(in_features=512, out_features=256, bias=True)
  (3): ELU(alpha=1.0)
  (4): Linear(in_features=256, out_features=128, bias=True)
  (5): ELU(alpha=1.0)
  (6): Linear(in_features=128, out_features=1, bias=True)
)
Error executing job with overrides: []
Traceback (most recent call last):
  File "/home/noah/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/utils/hydra.py", line 101, in hydra_main
    func(env_cfg, agent_cfg, *args, **kwargs)
  File "/home/noah/IsaacLab/scripts/reinforcement_learning/rsl_rl/train.py", line 167, in main
    runner = OnPolicyRunner(env, agent_cfg.to_dict(), log_dir=log_dir, device=agent_cfg.device)
  File "/home/noah/IsaacLab/_isaac_sim/kit/python/lib/python3.10/site-packages/rsl_rl/runners/on_policy_runner.py", line 48, in __init__
    raise ValueError("Observations for they key 'rnd_state' not found in infos['observations'].")
ValueError: Observations for they key 'rnd_state' not found in infos['observations'].
```

### System Info

Describe the characteristic of your environment:

- Commit: [7de6d6f](https://github.com/isaac-sim/IsaacLab/commit/7de6d6fef9424c95fc68dc767af67ffbe0da6832)
- Isaac Sim Version: 4.5.0-rc.36+release.19112.f59b3005.gl
- OS: Ubuntu 20.04
- GPU:  four NVIDIA RTX A4500
- CUDA: 12.3
- GPU Driver: 545.23.08

### Additional context

Obviously the first concern is being able to actually run rsl_rl/train.py, but if anyone has any idea why increasing the history_length from 2 to 3 resulted in a worse walking gait, or what I might have unknowingly done to cause this, any suggestions would be a huge help. I have compared the agent.yaml and env.yaml files and they are the same, excluding the history length. So I don't know what could be causing the walking gait to be worse.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

- [ ] Criteria 1: Able to run rsl_rl training without errors
- [ ] Criteria 2

RandomOakForest, 2025-04-22:
Thank you for posting this. Which version of Isaac Lab is this? Thanks.

NoahSTEMStuff, 2025-04-22:
I did ```./isaaclab.sh -i rsl_rl``` and that seemed to allow it to train. I guess for some reason after I pulled, even though my rsl_rl was not updated, I think it was 2.2 instead of 2.3, the code did not raise any errors letting me know this was the case nor install the updated version, which is weird, because I thought I saw code that was supposed to do that when I was looking through the repo trying to fix my issue. 

I'd like to also make sure that I have the correct versions that Isaac lab expects me to have, so there is no differences between my workspace and what has been tested. Is there anything else that has been updated since the end of February/beginning of March that I should update/reinstall?

RandomOakForest, 2025-04-23:
Great to hear you are training again. Regarding your last question, it is best to use the Isaac Lab releases instead of the ToT because releases undergo specific tests, while ToT is ongoing progress. If something like this eventually happens, the best is to try a new clean install to be sure the environment is well set up. I will close this report for now. Thanks for your interest in Isaac Lab.

closed 2025-04-23 by RandomOakForest