isaac-sim/IsaacLab #2350: [Bug Report] After training Isaac-Factory-PegInsert-Direct-v0, the .pth model cannot be converted to policy.onnx

mengkui-wang, 2025-04-23:
Hi, I found that the RL_games library does not export the onnx function in the play.py environment.
1、	RL-games Train 
Command：  `./isaaclab.sh -p `scripts/reinforcement_learning/rl_games/train.py` --task Isaac-Ant-v0 --headless`
2、	RL-games Play
Command：`./isaaclab.sh -p scripts/reinforcement_learning/rl_games/play.py --task Isaac-Ant-v0 --num_envs 32 --checkpoint /PATH/TO/model.pth`
```
=> loading checkpoint 'logs/rl_games/ant/2025-04-18_14-07-12/nn/ant.pth'
/miniconda3/envs/lab/lib/python3.10/site-packages/rl_games/algos_torch/torch_ext.py:62: FutureWarning: You are using `torch.load` with `weights_only=False` (the current default value), which uses the default pickle module implicitly. It is possible to construct malicious pickle data which will execute arbitrary code during unpickling (See https://github.com/pytorch/pytorch/blob/main/SECURITY.md#untrusted-models for more details). In a future release, the default value for `weights_only` will be flipped to `True`. This limits the functions that could be executed during unpickling. Arbitrary objects will no longer be allowed to be loaded via this mode unless they are explicitly allowlisted by the user via `torch.serialization.add_safe_globals`. We recommend you start setting `weights_only=True` for any use case where you don't have full control of the loaded file. Please open an issue on GitHub for any issues related to this experimental feature.
  return func(*args, **kwargs)
```
After checking the rsl_rl code, I found that RL_games did not implement the export policy.onnx code
```
# obtain the trained policy for inference
    policy = ppo_runner.get_inference_policy(device=env.unwrapped.device)

    # export policy to onnx/jit
    export_model_dir = os.path.join(os.path.dirname(resume_path), "exported")
    export_policy_as_jit(ppo_runner.alg.policy, ppo_runner.obs_normalizer, path=export_model_dir, filename="policy.pt")
    export_policy_as_onnx(
        ppo_runner.alg.policy, normalizer=ppo_runner.obs_normalizer, path=export_model_dir, filename="policy.onnx"
    )

```

RandomOakForest, 2025-04-23:
Thanks for posting this. Which version of Isaac Lab and Isaac Sim are you using?

mengkui-wang, 2025-04-24:
isaaclab 0.36.5
isaacsim 4.5.0.0

closed 2025-06-03 by RandomOakForest