isaac-sim/IsaacLab #2351: [Bug Report] [Error] [gpu.foundation.plugin] No carb::graphics::DescriptorSet for pool 1 on device 0

wolfcanli, 2025-04-23:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

Isaac sim: workstation manner installation. 
when launch isaac-sim.sh,
`./isaac-sim.sh`
 it reports error:
[Error] [gpu.foundation.plugin] No carb::graphics::DescriptorSet for pool 1 on device 0
### Steps to reproduce

Please try to provide a minimal example to reproduce the bug. Error messages and stack traces are also helpful.

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: 4.5
- OS: Ubuntu 22.04
- GPU: RTX A4000(16G)
- CUDA: 12.4
- GPU Driver: 570.133.07

### Additional context

Add any other context about the problem here.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

Ethereal1679, 2025-04-23:
what? But how can I solve it?

RandomOakForest, 2025-04-23:
Thanks for catching this. We will review and follow up.

RandomOakForest, 2025-04-25:
Following up on this, have you tried the [recommended driver](https://docs.isaacsim.omniverse.nvidia.com/4.5.0/installation/requirements.html#driver-requirements)?

openhe-hub, 2025-08-15:
I have the same problem of this error: "No carb::graphics::DescriptorSet for pool 1 on device 0"
My environment is:
* Isaac Sim Version: 4.5
* OS: Ubuntu 22.04
* GPU: RTX 4070
* CUDA: 12.8
* GPU Driver: 570.144

RandomOakForest, 2025-08-15:
Please post a new issue in the Isaac Sim repository.