isaac-sim/IsaacLab #2352: [Question] Result of root_physx_view.get_link_incoming_joint_force() is Incorrect in some cases

XuSen-Proto, 2025-04-23:
### Question

Hi, currently I’m facing an issue that I couldn’t fix. I tried to obtain the net wrench experienced at the float base, like what I have done in isaacgym by calling the ` refresh_force_sensor_tensor()` to obtain root wrench in the world. However, the incoming wrench at the root is `[0., 0., 0., 0., 0., 0.]` at the root of the kinematic tree in isaaclab. To cope with that case, I addded a parent body to the original root body and connnected then with a fixed joint. Then I defined the added body as the new root and read the incoming wrench at the original root. For a humanoid walking case, the force component in the z-axis is always minus, which seems to be unreal.

XuSen-Proto, 2025-04-25:
solved

closed 2025-04-25 by XuSen-Proto

RandomOakForest, 2025-04-25:
Great to hear that. Could you share your solution to have it here as a reference. It may help someone else. Thanks.