isaac-sim/IsaacLab #2354: Training Crash

jie-zh, 2025-04-23:
system configuration

- ubuntu: 22.04
- cuda: 12.4
- nvidia driver: 550.120
- isaaclab: 2.02
- isaacsim: 4.5.0

The following is the crash report, has someone met the same problem? At first, when i train about 10k or less iterations, the crash would happen. I have tried several methods to solve the problem according to the report, including 1:  changing the logging tool from tensorboard to neptune 2: Disabling the newest instruction set 3: transitioning ubuntu version to 20.04/24.04 , but neither of them works. Now the crash happens almost at the beginning of the training, at about less then 1k iterations.

The only valid solution to the problem is using windows instead of ubuntu. The training can normally run on windows, which proves that the hardware works well.    
```
Fatal Python error: Illegal instruction

![Image](https://github.com/user-attachments/assets/f1fc903a-edb2-4454-961c-f296999bfead)

Thread 0x00007641497fe6c0 (most recent call first):
<no Python frame>

Thread 0x0000764149fff6c0 (most recent call first):
File "/home/agile/miniforge3/envs/env_isaaclab/lib/python3.10/threading.py", line 324 in wait
File "/home/agile/miniforge3/envs/env_isaaclab/lib/python3.10/queue.py", line 180 in get
File "/home/agile/miniforge3/envs/env_isaaclab/lib/python3.10/site-packages/tensorboard/summary/writer/event_file_writer.py", line 269 in _run
File "/home/agile/miniforge3/envs/env_isaaclab/lib/python3.10/site-packages/tensorboard/summary/writer/event_file_writer.py", line 244 in run
File "/home/agile/miniforge3/envs/env_isaaclab/lib/python3.10/threading.py", line 1016 in _bootstrap_inner
File "/home/agile/miniforge3/envs/env_isaaclab/lib/python3.10/threading.py", line 973 in _bootstrap

Current thread 0x00007646979cf600 (most recent call first):
File "/home/agile/miniforge3/envs/env_isaaclab/lib/python3.10/site-packages/torch/optim/adam.py", line 613 in _multi_tensor_adam
File "/home/agile/miniforge3/envs/env_isaaclab/lib/python3.10/site-packages/torch/optim/adam.py", line 784 in adam
File "/home/agile/miniforge3/envs/env_isaaclab/lib/python3.10/site-packages/torch/optim/optimizer.py", line 154 in maybe_fallback
File "/home/agile/miniforge3/envs/env_isaaclab/lib/python3.10/site-packages/torch/optim/adam.py", line 223 in step
File "/home/agile/miniforge3/envs/env_isaaclab/lib/python3.10/site-packages/torch/optim/optimizer.py", line 91 in _use_grad
File "/home/agile/miniforge3/envs/env_isaaclab/lib/python3.10/site-packages/torch/optim/optimizer.py", line 487 in wrapper
File "/home/agile/miniforge3/envs/env_isaaclab/lib/python3.10/site-packages/rsl_rl/algorithms/ppo.py", line 386 in update
File "/home/agile/miniforge3/envs/env_isaaclab/lib/python3.10/site-packages/rsl_rl/runners/on_policy_runner.py", line 260 in learn
File "/home/agile/IsaacLab/scripts/reinforcement_learning/rsl_rl/train.py", line 183 in main
File "/home/agile/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/utils/hydra.py", line 101 in hydra_main
File "/home/agile/miniforge3/envs/env_isaaclab/lib/python3.10/site-packages/hydra/core/utils.py", line 186 in run_job
File "/home/agile/miniforge3/envs/env_isaaclab/lib/python3.10/site-packages/hydra/_internal/hydra.py", line 119 in run
File "/home/agile/miniforge3/envs/env_isaaclab/lib/python3.10/site-packages/hydra/_internal/utils.py", line 458 in <lambda>
File "/home/agile/miniforge3/envs/env_isaaclab/lib/python3.10/site-packages/hydra/_internal/utils.py", line 220 in run_and_report
File "/home/agile/miniforge3/envs/env_isaaclab/lib/python3.10/site-packages/hydra/_internal/utils.py", line 457 in _run_app
File "/home/agile/miniforge3/envs/env_isaaclab/lib/python3.10/site-packages/hydra/_internal/utils.py", line 394 in _run_hydra
File "/home/agile/miniforge3/envs/env_isaaclab/lib/python3.10/site-packages/hydra/main.py", line 94 in decorated_main
File "/home/agile/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/utils/hydra.py", line 104 in wrapper
File "/home/agile/IsaacLab/scripts/reinforcement_learning/rsl_rl/train.py", line 191 in <module>

Extension modules: numpy.core._multiarray_umath, numpy.core._multiarray_tests, numpy.linalg._umath_linalg, numpy.fft._pocketfft_internal, numpy.random._common, numpy.random.bit_generator, numpy.random._bounded_integers, numpy.random._mt19937, numpy.random.mtrand, numpy.random._philox, numpy.random._pcg64, numpy.random._sfc64, numpy.random._generator, scipy._lib._ccallback_c, scipy.sparse._sparsetools, _csparsetools, scipy.sparse._csparsetools, scipy.linalg._fblas, scipy.linalg._flapack, scipy.linalg.cython_lapack, scipy.linalg._cythonized_array_utils, scipy.linalg._solve_toeplitz, scipy.linalg._decomp_lu_cython, scipy.linalg._matfuncs_sqrtm_triu, scipy.linalg._matfuncs_expm, scipy.linalg._linalg_pythran, scipy.linalg.cython_blas, scipy.linalg._decomp_update, scipy.sparse.linalg._dsolve._superlu, scipy.sparse.linalg._eigen.arpack._arpack, scipy.sparse.linalg._propack._spropack, scipy.sparse.linalg._propack._dpropack, scipy.sparse.linalg._propack._cpropack, scipy.sparse.linalg._propack._zpropack, scipy.sparse.csgraph._tools, scipy.sparse.csgraph._shortest_path, scipy.sparse.csgraph._traversal, scipy.sparse.csgraph._min_spanning_tree, scipy.sparse.csgraph._flow, scipy.sparse.csgraph._matching, scipy.sparse.csgraph._reordering, scipy.spatial._ckdtree, scipy._lib.messagestream, scipy.spatial._qhull, scipy.spatial._voronoi, scipy.spatial._distance_wrap, scipy.spatial._hausdorff, scipy.special._ufuncs_cxx, scipy.special._ufuncs, scipy.special._specfun, scipy.special._comb, scipy.special._ellip_harm_2, scipy.spatial.transform._rotation, torch._C, torch._C._dynamo.autograd_compiler, torch._C._dynamo.eval_frame, torch._C._dynamo.guards, torch._C._dynamo.utils, torch._C._fft, torch._C._linalg, torch._C._nested, torch._C._nn, torch._C._sparse, torch._C._special, hid, PIL._imaging, scip
```

[image]

RandomOakForest, 2025-04-25:
Thank you for posting this. Could you please try the [recommended driver](https://docs.isaacsim.omniverse.nvidia.com/4.5.0/installation/requirements.html#driver-requirements)? also, could you share what are you running from the terminal?

jie-zh, 2025-04-27:
> Thank you for posting this. Could you please try the [recommended driver](https://docs.isaacsim.omniverse.nvidia.com/4.5.0/installation/requirements.html#driver-requirements)? also, could you share what are you running from the terminal?

Thank for the response. I have also tried driver 550.144 and also 560, as expressed in document the latest driver is recommended.  The running command from the terminal is common like this:  "./isaaclab.sh -p scripts/reinforcement_learning/rsl_rl/train.py --task Isaac-Velocity-Flat-G1-v0 --headless" . Some changes about rewards are made in the environment.

orangelee89, 2025-05-19:
hi , I have similar issue as yours ,  do you solve this problem?

RandomOakForest, 2025-06-03:
Following up, if you still face this issue with the 535.129.03 driver, please open a new issue as a _bug report_. I will close this issue for now. Thank you for your interest in Isaac Lab.

closed 2025-06-03 by RandomOakForest