isaac-sim/IsaacLab #2355: [Question] Torque control robotic arm

luju97, 2025-04-23:
### Question

Hi,

I would like to train the Franka robot to loft an objects, same as in the [example](https://isaac-sim.github.io/IsaacLab/main/source/overview/environments.html) (Franka Isaac-Lift-Cube-Franka-v0). 
I would like to use force control `JointEffortActionCfg` rather than position control `JointPositionActionCfg`..

What I did:
- I used `JointEffortActionCfg` for all of the joints of the robot:
```python
self.actions.arm_action = mdp.JointEffortActionCfg(
            asset_name="robot",
            joint_names=["panda_joint.*"], 
            scale=1.0,       
            offset=0.0,
        )
```
- I have enabled gravity compensation.
- I have set the damping and the stiffness to zero:
```python
"panda_shoulder": ImplicitActuatorCfg(
    joint_names_expr=["panda_joint[1-4]"],
    effort_limit=87.0,
    velocity_limit=2.175,
    stiffness=0.0,
    damping=0.0,
),
"panda_forearm": ImplicitActuatorCfg(
    joint_names_expr=["panda_joint[5-7]"],
    effort_limit=12.0,
    velocity_limit=2.61,
    stiffness=0.0,
    damping=0.0,
),
```

However, I cannot train the robot effectively to lift the cube. Why is it? What am I doing wrong?

Thank you in advance!

RandomOakForest, 2025-04-25:
Thanks for posting this. Could you elaborate on what you are seeing and what you mean by "effectively" in this lift task? Thanks.

closed 2025-05-02 by luju97

fl1ps1de5, 2025-05-14:
Was this ever answered? I am trying to do something but using velocity actions and would like some guidance - not sure how to devise a suitable config.