isaac-sim/IsaacLab #2361: [Question] Tiled Camera Keeps Accumulating Frames in VRAM

tkzemrhn, 2025-04-23:
I'm using the latest IsaacLab version, alongside IsaacSim 4.5 on a system 16GB of VRAM. I'm currently working on training an RL agent to fly in a cluttered environments using cameras using the OmniDrones library (and its port to IsaacSim 4.5). For the camera, I'm using TiledCamera with distance_to_image_plane, and a resolution of 80x60. As the sim runs, the VRAM usage keeps increasing, it doesn't hold steady or decrease at any point. I added my environment code below in the env file, I'm not sure if there's a clear cache sort of function that I missed, or should I re-initialize the cameras or what. I also added the WandB output to show what's going on:

[image]

[env_file.txt](https://github.com/user-attachments/files/19875456/env_file.txt)

RandomOakForest, 2025-04-24:
Thank you for posting this. How many GPUs are you using? Could you share details about your platform? Thanks.

tkzemrhn, 2025-04-24:
Hi,
I'm using a single desktop version RTX 4080 with 16GB VRAM, a Ryzen7 5800x3D alongside 32GB of system RAM. The GPU is running "Driver Version: 560.35.05", "CUDA Version: 12.6". The conda environment has torchrl-nightly, tensordict-nightly, and PyTorch 2.7.0, though I don't think the library versions are the cause of the issue.

RandomOakForest, 2025-04-25:
Thanks! Do you see a similar increase in GPU memory allocated when running tutorials, such as the carpool camera environment?

giangdao1402, 2025-04-26:
DId you solve this problem, i encounter a same problem when using camera for RL @tkzemrhn

closed 2025-06-03 by RandomOakForest