isaac-sim/IsaacLab #2366: [Bug Report] ImportError: cannot import name 'RslRlRndCfg' from 'isaaclab_rl.rsl_rl'

MARMOTatZJU, 2025-04-24:
### Describe the bug

#### Background

I am working on [LeggedLab](https://github.com/Hellod035/LeggedLab) which is built on top of _IsaacLab_. In my runtime environment, my IsaacLab's version is **v2.0.2** which is installed through _pip_. The latest commit of LeggedLab depends on the _nightly version_ of IsaacLab (which should be [Commit 21f7136](https://github.com/isaac-sim/IsaacLab/commit/21f7136325136ca3f6ca4e0a8125edffe5c24f7e)) and I am trying to move to this latest commit.

#### Problem

However, an ImportError occurs when I try to run the code after rebasing.

### Steps to reproduce

Here I post my log which contains the related code as well as the detailed error info.:

```text
"***/legged_lab/legged_lab/envs/base/base_env_config.py", line 11, in <module>
    from isaaclab_rl.rsl_rl import RslRlOnPolicyRunnerCfg, RslRlPpoActorCriticCfg, RslRlPpoAlgorithmCfg, RslRlRndCfg, RslRlSymmetryCfg  # noqa:F401
ImportError: cannot import name 'RslRlRndCfg' from 'isaaclab_rl.rsl_rl' (***/python3.10/site-packages/isaaclab/source/isaaclab_rl/isaaclab_rl/rsl_rl/__init__.py)、
```

### My personal suggestion

I understand that this issue is due to the changed interface of IsaacLab between v2.0.2 and the latest commt ([Commit 21f7136](https://github.com/isaac-sim/IsaacLab/commit/21f7136325136ca3f6ca4e0a8125edffe5c24f7e)).

Personally, I prefer to work on a tagged commit (with a version number) of IsaacLab in order to have a stable interface. Seeing that backward-incompatible changes are already made to IsaacLab, I recommend to tag a new version of IsaacLab as an official version after comprehensive testing of the code.

### System Info

Describe the characteristic of your environment:

- Commit: 21f7136
- Isaac Sim Version: 4.5.0, installed through pip
- OS: Ubuntu 22.04
- GPU: RTX 4090 (24GB RAM version)
- CUDA: 12.8
- GPU Driver: 535.183.01

### Checklist

- [X] I have checked that there is no similar issue in the repo (**required**)
- [X] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1: a new version is tagged on the latest commit after comprehensive testing of the code.
- [ ] Criteria 2

RandomOakForest, 2025-04-24:
Thank you for posting this. Could you share the code until the `import` giving you trouble? Are you sure it is called after launching the app?

MARMOTatZJU, 2025-04-24:
> Thank you for posting this. Could you share the code until the `import` giving you trouble? Are you sure it is called after launching the app? 

Thanks for your attention. I will try to upload the code tomorrow.

By the way for now, I conjecture that this issue would be purely related to the version/commit. So in the limit I wish that an up-to-date version can be tagged so that I can move on to that version on which the latest commit of Legged Lab should run without error.

Could you give me any clue on the date where the next version will be tagged/released (e.g. v2.0.3)?

MARMOTatZJU, 2025-04-27:
@RandomOakForest That's awesome! I am going test `v2,1,0` today. I will come back with the result.

MARMOTatZJU, 2025-04-28:
@RandomOakForest Case closed. `v2.1.0` works. Many thanks to you and your team!

RandomOakForest, 2025-06-03:
Following up, there are internal tests that are also performed. I will close this issue for now. Thank you for your interest in Isaac Lab.