isaac-sim/IsaacLab #2369: [Question] Clarification on joint_drive vs. actuator gains

KyleM73, 2025-04-24:
### Question

in the URDF importer, actuators can be configured with stiffness and damping. the importer also requires a joint_drive configuration which includes a stiffness and damping in PD mode. what is the physical meaning of the joint drive? it seems people have asked on the isaacsim forms but no clear answer is given. should the joint drive be configured with the same gains as the actuators, or do they correspond to different things? what value is set if None is passed for both the stiffness and damping? any additional information or documentation would be hugely appreciated, as the current documentation is not very clear on what the joint_drive is doing.

RandomOakForest, 2025-04-25:
Thanks for posting this. _Joint drives apply a force to the joint in order to maintain a position and/or a velocity_, you can find the [details here](https://docs.omniverse.nvidia.com/kit/docs/omni_physics/latest/dev_guide/rigid_bodies_articulations/joints.html#joint-drive). The docs for Isaac Sim 4.2.0 describe about [tunning joint drive gains here](https://docs.omniverse.nvidia.com/isaacsim/latest/advanced_tutorials/tutorial_advanced_joint_tuning.html#understanding-joint-drives). It is not clear why Isaac Sim 4.5.0 docs do not lead to a similar page. And there is something about the URDF importer that needs to be more clear. I will review with the Isaac Lab and Isaac Sim teams to provide more details soon. Thanks for your interest in Isaac Lab.

KyleM73, 2025-04-29:
To me this sounds like exactly what the actuators do when the policy has a position or velocity action space. In the legged locomotion examples, the action space of the policy is a position offset on some default (usually stance) configuration. For explicit actuators, this produces the control law
`tau = Kp*(q_default + Ka*a - q) - Kd*dq`
where Kp, Kd, and Ka are the stiffness, damping, and scale of the policy actions, respectively. This torque is computed explicitly.

It is still unclear to me how the joint drive is factored into this setup:
1) Are joint drives only used if the actuators are of implicit type? In which case the gains of the joint drives should match the stiffness and damping of the configured actuators?
2) But in explicit mode, are the joint drives being used at all? 

Thank you for the clarification.

vicltz, 2025-05-27:
https://github.com/isaac-sim/IsaacLab/blob/3b6d615f9aff7435fdafaa75a0d59365500a428c/source/isaaclab/isaaclab/assets/articulation/articulation.py#L1371C1-L1380C87

I think this code answer this question:
As far as I understand: 

- In case of implicit actuator, actuators gains are written to joint gains and the underlying physx joint mechanism is then used to actuate the model.
- In the case of explicit actuators, those joint gains are set to 0 to avoid interference and the computed torque is computed above that by the explicit actuator.

In both cases, the joint gains defined in the model before adding the actuators have no effect. @RandomOakForest  Can you please confirm or infirm this analysis ?

RandomOakForest, 2025-06-04:
Thanks for following up. Here is a summary that may address your comments above. I will move this post to our _Discussions_ section for follow up. 

In NVIDIA Isaac Lab and Isaac Sim, and thus the URDF importer, **joint drives** and **actuators** are somewhat distinct components with different roles in joint control. 

#### **1. Joint Drive Definition**

- **Physical meaning**:
Joint drives represent the physics solver's internal PD controller for joints, governed by:
```math
\text{Effort} = \text{stiffness} \cdot (\text{position error}) + \text{damping} \cdot (\text{velocity error})
```
- where
  - **Stiffness**: Proportional gain for position error (units: Nm/rad for revolute joints, N/m for prismatic) [^3][^6].
  - **Damping**: Derivative gain for velocity error (units: Nm/(rad/s) or N/(m/s)) [^3][^6].
  - Operates at the physics engine level, integrating constraints directly into solver calculations [^4][^9].

#### **2. Actuator Interactions**

- **Explicit vs. Implicit Actuators**:
    - **Explicit** (e.g., DC motors): Compute efforts independently, ignoring joint drive parameters [^4][^5].
    - **Implicit**: Rely on joint drive's PD controller for effort computation [^4][^8].
- **Configuration Guidelines**:
    - For **position control**: Set high stiffness (e.g., $10^6$) and moderate damping (e.g., $10^3$) [^2][^6].
    - For **velocity control**: Use damping as primary gain with stiffness set to zero [^2][^9].
    - For **explicit actuators**: Disable joint drives (stiffness/damping = 0) [^5][^9].

#### **3. Default Behavior with `None` Values**

When `stiffness` and `damping` are unspecified (`None`):

1. **URDF values** take precedence if defined in `<dynamics>` tags [^2][^10].
2. **Fallback** to `0.0` if no URDF parameters exist [^3][^10].

#### **4. Joint Drive Usage with Explicit Actuators**

| Component | Role in Explicit Actuators | Default Behavior (No Configuration) |
| :-- | :-- | :-- |
| **Joint Drive** | Physics engine’s internal PD controller | Disabled (stiffness/damping = 0.0) |
| **Actuator** | Explicit torque computation via control law | Full control over torque output |

- **(1) Joint Drives + Implicit Actuators**:
Joint drives **only** apply when using *implicit actuators* (physics-engine PD control). In this case, joint drive gains **must match** the actuator’s configured stiffness/damping.
- **(2) Explicit Actuators**:
Joint drives are **disabled** (stiffness/damping = 0.0) because the actuator computes torque independently.

#### **5. Code in `articulation.py`**

```python
if self._actuator_root.has_actuators:
    # Write actuator gains to sim for implicit actuators
    if self._actuator_root.is_implicit():
        self._write_joint_gains_to_sim()
    else:
        # Explicit actuators: Zero out joint gains
        self._write_joint_gains_to_sim(stiffness=0.0, damping=0.0)
```

1. **Implicit Actuators**:
    - `self._write_joint_gains_to_sim()` transfers the actuator's configured `stiffness` and `damping` to the PhysX joint's internal PD controller.
    - The physics engine then uses these gains to compute joint efforts
    - Pre-existing joint gains (from URDF or other configurations) **are overwritten** by actuator parameters.
2. **Explicit Actuators**:
    - `stiffness=0.0` and `damping=0.0` are explicitly set, disabling the PhysX joint's internal PD controller.
    - The actuator computes torque independently (e.g., via motor models or custom control laws) and applies it directly.
    - Any pre-configured joint gains **are ignored** because the physics engine's PD terms are zeroed out.

In summary:

- **Implicit Actuators**: Actuator gains → PhysX joint gains.
- **Explicit Actuators**: PhysX joint gains → `0.0`, actuator torque dominates.
- **Pre-existing joint gains**: Always overridden by actuator configuration.

For implementation details, please refer to Isaac Lab's [actuator documentation](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.actuators.html) and [joint tuning guide](https://docs.omniverse.nvidia.com/isaacsim/latest/advanced_tutorials/tutorial_advanced_joint_tuning.html).

*References*

[^2]: https://docs.omniverse.nvidia.com/isaacsim/latest/features/environment_setup/ext_omni_isaac_urdf.html

[^3]: https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.sim.schemas.html

[^4]: https://isaac-sim.github.io/IsaacLab/main/source/overview/core-concepts/actuators.html

[^5]: https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.actuators.html

[^6]: https://nvidia-omniverse.github.io/PhysX/physx/5.1.3/docs/Articulations.html

[^7]: https://docs.omniverse.nvidia.com/isaacsim/latest/isaac_lab_tutorials/tutorial_policy_deployment.html

[^8]: https://isaac-sim.github.io/IsaacLab/main/source/tutorials/01_assets/add_new_robot.html

[^9]: https://docs.isaacsim.omniverse.nvidia.com/4.5.0/robot_setup/import_urdf.html

[^10]: https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.sim.converters.html