isaac-sim/IsaacLab #2371: [Question] Domain Randomziation

Entongsu, 2025-04-24:
Hi,

I'm working on reinforcement learning training in Isaac Sim and would appreciate your help with two questions:

Object Scaling: I’d like to perform domain randomization on the scale of objects, but I haven’t found the relevant functions or documentation for this. Could you point me in the right direction?

Object Randomization Across Environments: I’d also like to randomize the objects used in training. From what I’ve seen, Isaac Sim requires each environment instance to load the same number of objects. Is it possible to configure different environments to each load a single but different object, rather than all loading the same set?

Thanks in advance for your help!

RandomOakForest, 2025-04-25:
Thanks for posting this. Have you reviewed [this tutorial on domain randomization](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/03_envs/create_direct_rl_env.html#domain-randomization)? This would be a place to start. Thank you for your interest in Isaac Lab.

torydebra, 2025-05-05:
Regarding the first question, I am using functions like `randomize_visual_color`, `randomize_rigid_body_mass`, `randomize_rigid_body_scale` `randomize_rigid_body_material` `randomize_rigid_body_material` `randomize_visual_texture_material`

But I was interested in the second question as well. Which is the best way to randomize the shape of the object itself? For example, from cubes to cylinders? Is it possible?

RandomOakForest, 2025-06-04:
Thanks for following up. I will move this post to our _Discussions_ section for follow up. In the meantime, here is a summary of what to consider.

To randomize object shapes (e.g., switching between cubes and cylinders), use **USD asset swapping** combined with environment-specific spawning:

### Shape Randomization Implementation

**1. Multi-Shape Asset Configuration**

```python
from isaaclab.assets import AssetBaseCfg

SHAPE_USD_PATHS = [
    "/Isaac/Props/Shapes/Cube.usd",
    "/Isaac/Props/Shapes/Cylinder.usd",
    "/Isaac/Props/Shapes/Sphere.usd"
]

object_cfg = AssetBaseCfg(
    prim_path="/World/Env_.*/Object",
    spawn=usd_utils.spawn_rigid_objects(
        prim_paths=[f"/World/Env_{i}/Object" for i in num_envs],
        usd_paths=np.random.choice(SHAPE_USD_PATHS, num_envs)
    )
)
```

**2. Per-Environment Variation**
Each environment instance receives a randomly selected shape while maintaining:

- Physics collision properties through unified API access
- Consistent material/texture randomization pipelines
- Positional constraints across different geometries

### Key Requirements

1. **Pre-built USD assets** for each shape variant
2. **Matching collision meshes** across different geometries
3. **Consistent prim hierarchy** for transform operations

### Implementation Considerations

- Use `GridCloner` for efficient multi-environment spawning [^4]
- Combine with scale randomization through `actor_params.scale.range` [^3]
- For procedural generation, implement mesh deformation using Omniverse Mesh API as shown in the vertex modification approach [^1]

An example workflow from the NVIDIA forums shows successful shape randomization by:

1. Generating multiple USD files with different geometries
2. Randomly selecting them during environment initialization
3. Using rigid body aggregation for physics handling [^1]

This approach enables true geometric diversity while maintaining Isaac Lab's optimized physics pipeline.

_References_

[^1]: https://forums.developer.nvidia.com/t/domain-randomization-of-an-object-shape/214426

[^2]: https://docs.omniverse.nvidia.com/isaacsim/latest/gui_tutorials/tutorial_intro_simple_objects.html

[^3]: https://isaac-sim.github.io/IsaacLab/main/source/migration/migrating_from_omniisaacgymenvs.html

[^4]: https://docs.omniverse.nvidia.com/isaacsim/latest/replicator_tutorials/tutorial_replicator_isaac_randomizers.html