isaac-sim/IsaacLab #2375: [Question] How to get the segmentation of each robot link?

DuYi-angle, 2025-04-25:
### Question

Here is my robot cfg, currently I can only get the semantic label of the robot as a whole. I would like to know how I can get the semantic label of each robot link as [https://github.com/isaac-sim/IsaacLab/issues/809](url).

`robot = ArticulationCfg(
        prim_path="/World/envs/env_.*/Robot",
        spawn=sim_utils.UsdFileCfg(
            usd_path="/home/zhangsl/TeleAISim/teleaisim/assets/humanoid/configuration/h1_2_xhand.usd",
            activate_contact_sensors=True,
            rigid_props=sim_utils.RigidBodyPropertiesCfg(
                rigid_body_enabled=True,
                disable_gravity=False,
                # kinematic_enabled=True,
                retain_accelerations=False,
                linear_damping=0.0,
                angular_damping=0.0,
                max_linear_velocity=1000.0,
                max_angular_velocity=1000.0,
                max_depenetration_velocity=1,
            ),
            articulation_props=sim_utils.ArticulationRootPropertiesCfg(
                enabled_self_collisions=True, solver_position_iteration_count=4, solver_velocity_iteration_count=4
            ),
            collision_props=sim_utils.CollisionPropertiesCfg(collision_enabled=True, contact_offset=0.005, rest_offset=0.0),
            semantic_tags=[("class", "robot"), ("color", "white")],
        ),`

RandomOakForest, 2025-04-25:
Thank you for posting this. That may be done using [Replicator](https://docs.omniverse.nvidia.com/extensions/latest/ext_replicator.html) after adding a camera sensor to the scene. Is this what you are referring to? Can you elaborate on an example data output you expect having the semantic label of interest?

DuYi-angle, 2025-04-27:
I found that "instance_id_segmentation_fast" was the rendering mode I needed. I had mistakenly used "semantic_segmentation" before.

closed 2025-04-27 by DuYi-angle