isaac-sim/IsaacLab #2378: [Proposal] Add support for spawning meshes from obj and stl files

han-xudong, 2025-04-25:
### Proposal

Adding support for spawning meshes from .obj and .stl files.

### Motivation

When I was trying to spawn a mesh from an STL file, I found that only several kinds of geometries were supported. And I also found that these geometries were generated by `trimesh` which can load meshes from OBJ and STL files. Therefore, it's available to create a function based on `trimesh` to make it work.

### Alternatives

- Add `spawn_mesh_from_file` in `source/isaaclab/isaaclab/sim/spawners/meshes/meshes.py`
- Add `MeshFileCfg` in `source/isaaclab/isaaclab/sim/spawners/meshes/meshes_cfg.py`
- Update `source/isaaclab/isaaclab/sim/spawners/meshes/__init__.py`

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Support spawning meshes from OBJ and STL files.

RandomOakForest, 2025-04-26:
Thank you for posting this proposal. The team will review it.

han-xudong, 2025-05-06:
@RandomOakForest Is there any updates? I think the PR is quite clear.

AntoineRichard, 2026-08-27:
Appreciate the proposal and the prototype, @han-xudong — direct OBJ/STL spawning is a reasonable thing to want.

We're going to keep a single mesh-loading path, so we won't add OBJ/STL spawners to `meshes_cfg.py` (which stays procedural-primitives only). The supported route is [`MeshConverterCfg`](https://github.com/isaac-sim/IsaacLab/blob/develop/source/isaaclab/isaaclab/sim/converters/mesh_converter.py) or `scripts/tools/convert_mesh.py` to produce a USD, then spawn it with `sim_utils.UsdFileCfg`; conversion is cached by config hash, so repeat runs are cheap, and you get control over collision approximation and instancing. PR #2379 was closed by you, so nothing is pending there. Reopen if the converter route leaves a real gap.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard