isaac-sim/IsaacLab #2381: [Bug Report] LFS Files Not Found

asuzukii, 2025-04-25:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

When trying to rebase my fork, some of the LFS files that have been added recently fails the fetch.

### Steps to reproduce

I have a 2.0.2 Fork and I attempt to rebase on the new 2.1 release. Steps to run:
```bash
git remote add upstream https://github.com/isaac-sim/IsaacLab

# Fetch all the branches of that remote into remote-tracking branches
git fetch upstream

git checkout origin/main

# Rewrite your main branch so that any commits of yours that aren't already in
git rebase upstream/main 
```

Error I get is:
```
Downloading docs/source/_static/setup/cloudxr_bimanual_teleop.gif (4.2 MB)
Error downloading object: docs/source/_static/setup/cloudxr_bimanual_teleop.gif (7b09a38): Smudge error: Error downloading docs/source/_static/setup/cloudxr_bimanual_teleop.gif (7b09a38a99ed25a17c1e1e36abf62b228807677e30c7fdb8ebe53b79f8d99551): [7b09a38a99ed25a17c1e1e36abf62b228807677e30c7fdb8ebe53b79f8d99551] Object does not exist on the server: [404] Object does not exist on the server
```

A quick google search seems like you guys could try something like this: https://stackoverflow.com/a/62855597

### System Info

Describe the characteristic of your environment:

- Commit: f3c5404
- Isaac Sim Version: 4.5.0
- OS:  Ubuntu 22.04
- GPU: Quadro RTX 4000
- CUDA: 12.4
- GPU Driver: 550.120

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] I can properly rebase on some commit after the 2.1 release commit

RandomOakForest, 2025-04-26:
Thank you for posting this. It is typically an LFS setup issue or a temporary git issue. Disabling `git-lfs`, pulling, then re-enabling it, may help. If it doesn't, try adding the following to your `~/.gitconfig`, then rebase again:
```
[filter "lfs"]
        clean = git-lfs clean -- %f
        smudge = git-lfs smudge -- %f
        process = git-lfs filter-process
        required = true
```
If that doesn't work, it may be just a temporary git issue, as you may have tried to clone while git is syncing to your forked depot.

asuzukii, 2025-05-07:
Hi, yes something with git was just not synced. It has since resolved. Thank you!

closed 2025-06-04 by RandomOakForest