isaac-sim/IsaacLab #2387: [Proposal] Google Colab Examples in Docs

j3soon, 2025-04-27:
### Proposal

Enable users to run Isaac Lab demos directly on Google Colab, without the need for local installation.

### Motivation

This could lower the entry barrier for those who are still hesitant to use Isaac Lab or who have not yet set up their own environment. In addition, this would be great for workshops or courses that lack the required hardware for all participants.

For reference, MuJoCo have a collection of [IPython notebooks](https://github.com/google-deepmind/mujoco?tab=readme-ov-file#getting-started) that can be directly executed on Google Colab, which is quite nice for beginners.

### Alternatives

Other free services similar to Google Colab

### Additional context

I've created [a notebook demo](https://colab.research.google.com/github/j3soon/isaac-sim-colab/blob/main/notebooks/isaac-lab-2.1.0-colab.ipynb) for the Cartpole & Ant environment, including training, recording video, and plotting learning curves. The source code can be found here: <https://github.com/j3soon/isaac-sim-colab>.

However, since this requires several hacks to run smoothly on Colab, we should tag this example as experimental if we decide to include it.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Google Colab Notebooks added to repository & docs.

RandomOakForest, 2025-04-28:
Thanks for posting this proposal. The team is constantly working on improving Isaac Lab, and having this type of tutorials is definitely on our radar. The team will review this.

StafaH, 2026-10-11:
Added some initial notebooks in #7876 to help new users! We will continue to update the notebooks in future isaaclab releases

closed 2026-10-11 by StafaH