isaac-sim/IsaacLab #2388: How to change the sub-object texture in a loaded usd file.

kaijieshi7, 2025-04-27:
Hi, I load the usd file as following

room = RigidObjectCfg( 
        prim_path="{ENV_REGEX_NS}/room",
        spawn = sim_utils.UsdFileCfg(
            usd_path=f"/home/kaijie/Desktop/one_room.usd",    
            rigid_props=sim_utils.RigidBodyPropertiesCfg(
                kinematic_enabled=True
            ),
        ),
        init_state=RigidObjectCfg.InitialStateCfg(pos=(-9.5, -11, 0)),
    ) 

Then, the I can get the room by 'room=env.scene['room']'.
My aim is to change the texture of table in the room. The table path is like 'room/table'.
How can I achieve this?

RandomOakForest, 2025-04-28:
Thanks for posting this. Have you reviewed this [Isaac Sim tutorial](https://docs.omniverse.nvidia.com/isaacsim/latest/how_to_guides/environment_setup.html#creating-modifying-assigning-materials)? By loading the room in Isaac Sim, you should be able to see the _prim path_ of the table as you select the table of interest. Let us know if  you need further help.

kaijieshi7, 2025-04-29:
I use it, and get error: 'xxx' was deleted while being used by a shape in a tensor view class. The physics.tensors simulationView was invalidated.
2025-04-28 23:17:30 [34,630ms] [Error] [omni.physx.tensors.plugin] Simulation view object is invalidated and cannot be used again to call setDofActuationForces

RandomOakForest, 2025-06-04:
Following up, you may be landing in a couple of issues. Avoid deleting prims mid-simulation unless absolutely necessary. If deletion is required, fully reset the simulation context afterward:
```python
sim.reset()  # Reinitializes the simulation view
```
Also, holding references to PhysxView objects (e.g., robot.root_physx_view) after a scene reset or prim modification may cause a similar error. In that case, you will need to reacquire simulation views after structural scene changes:
```python
# After prim deletion/reset
self._robot = self._scene.actors["robot"]
self._physx_view = self._robot.root_physx_view  # Refresh reference
```

I'll move this post to our _Discussions_ section for following up.