isaac-sim/IsaacLab #2390: [Proposal] Error response from daemon: unknown or invalid runtime name: nvidia

DigitLion, 2025-04-28:
**Proposal:**  
Update Docker configuration to use modern GPU resource allocation methods instead of deprecated `runtime: nvidia` specification in CloudXR-related compose files.

**Motivation:**  
The current configuration leads to "unknown or invalid runtime name: nvidia" errors when starting containers, despite having proper NVIDIA toolchain installed. This occurs because:

1. Modern Docker versions (19.03+) prefer `deploy.resources` over explicit runtime specification
2. Mixed configuration approaches cause conflicts in GPU resource allocation
3. The error interrupts CloudXR runtime initialization, blocking development workflows

Core use cases affected:
- Local development/testing of CloudXR applications
- Isaac Sim integration with CloudXR services
- Multi-container GPU resource sharing scenarios

Related to frustration when:  
"I'm unable to launch the container environment even after properly configuring NVIDIA Container Toolkit and verifying GPU accessibility through standard Docker commands."

**Alternatives Considered:**
1. Maintaining legacy `runtime: nvidia` specification
   - Would require users to modify system-wide Docker configurations
   - Creates compatibility issues across different Docker versions
2. Using `--gpus` CLI flag instead of compose file configuration
   - Not scalable for multi-service environments
   - Harder to maintain in version-controlled configurations

**Additional Context:**  
Current error output:
```
Error response from daemon: unknown or invalid runtime name: nvidia
```

Successful GPU verification test:
```bash
docker run --rm --gpus all nvidia/cuda:11.0-base nvidia-smi
```
(Works correctly, proving base NVIDIA configuration is valid)

**Checklist:**
- [ ] I have checked that there is no similar issue in the repo

**Acceptance Criteria:**
- [ ] Updated compose files work without runtime specification errors
- [ ] Both CloudXR runtime and Isaac Lab containers can access GPU resources
- [ ] Configuration remains compatible with Docker 19.03+ and NVIDIA Container Toolkit 1.0+
- [ ] Documentation reflects any required changes in setup procedures

**System Information:**
<please fill with your actual system specs>
```
Docker Version: 28.0.4
NVIDIA Driver Version: 570.124.06
OS Version: Ubuntu 22.04.5 LTS
```

RandomOakForest, 2025-04-28:
Thank you for posting this proposal. The team will review it.

yanziz-nvidia, 2025-04-30:
Hi @DigitLion , thanks for the proposal. It makes sense to use `--gpus` over `runtime:nvidia` but the key is to make sure CloudXR Runtime and Isaac Lab share use of the same GPUs.

If you are using the docker python scripts with docker compose, you can edit [docker/docker-compose.cloudxr-runtime.patch.yaml](https://github.com/isaac-sim/IsaacLab/blob/main/docker/docker-compose.cloudxr-runtime.patch.yaml) file to remove `runtime: nvidia` and add the following to both `cloudxr-runtime` and `isaac-lab-base`:

```yaml
...
    deploy:
      resources:
        reservations:
          devices:
            - driver: nvidia
              count: all
              capabilities: [gpu]
...
```

If you use `device_ids` as mentioned in https://docs.docker.com/compose/how-tos/gpu-support/, please make sure you pass the same IDs for both containers.

I will send out a PR soon about the change needed.

PR #2411 (2025-05-03): Uses --gpus instead of nvidia runtime for docker
# Description

Use `--gpus all` instead of `nvidia: runtime` to prevent issues on unknown or invalid runtime name when running with docker or docker compose (python script).

Fixes #2390 

## Type of change

- Bug fix (non-breaking change which fixes an issue)
- This change requires a documentation update

## Screenshots

N/A

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with `./isaaclab.sh --format`
- [x] I have made corresponding changes to the documentation
- [x] My changes generate no new warnings
- [ ] I [... cut]
commit 8a913d09 (2025-05-03): Uses --gpus instead of nvidia runtime for docker (#2411)

# Description

Use `--gpus all` instead of `nvidia: runtime` to prevent issues on
unknown or invalid runtime name when running with docker or docker
compose (python script).

Fixes #2390 

## Type of change

- Bug fix (non-breaking change which fixes an issue)
- This change requires a documentation update

## Screenshots

N/A

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with
`./isaaclab.sh --format`
- [x] I have made corresponding changes to the documentation
- [x] My changes generate no new warning [... cut]
closed 2025-05-03 by kellyguo11