isaac-sim/IsaacLab #2391: [Bug Report] Step reward retains stale values when weight is dynamically set back to zero

bikcrum, 2025-04-28:
### Describe the bug

In the reward computation logic (`compute` function), when a reward term has a weight of zero, the code skips computation and does not update `self._step_reward` for that term.  
This is not an issue if a term's weight remains zero from initialization or is changed from zero to non-zero after the reward manager is created.  
However, if a term's weight is first changed to non-zero and later back to zero, the `self._step_reward` would retain stale (nonzero) values from earlier updates, since the loop skips updating it when the weight is zero.

Explicitly setting `self._step_reward` to zero when the weight is zero ensures correctness and prevents stale data,

### Steps to reproduce

1. Create or initialize a reward manager with a reward term having zero weight.
2. Change the weight of the term dynamically to a non-zero value during runtime.
3. Compute the reward and observe that `self._step_reward` is correctly updated with nonzero values.
4. Change the term’s weight back to zero dynamically.
5. Compute the reward again.
6. Observe that `self._step_reward` still retains the previous nonzero value because the update is skipped when the weight is zero.

Example (pseudo-code):

```python
# Assume reward manager initialized normally
reward_manager._term_cfgs[idx].weight = 2.0  # Set to non-zero
reward_manager.compute(dt=0.02)

print(reward_manager._step_reward[:, idx])  # --> Should show nonzero value

reward_manager._term_cfgs[idx].weight = 0.0  # Set back to zero
reward_manager.compute(dt=0.02)

print(reward_manager._step_reward[:, idx])  
# Expected: 0.0
# Observed: still showing stale nonzero value
```

### System Info

Describe the characteristic of your environment:

- Commit: 2e6946afb9b26f6949d4b1fd0a00e9f4ef733fcc
- Isaac Sim Version: 2.1.0
- OS: Ubuntu 22.04
- GPU: RTX 3060
- CUDA: 12.4
- GPU Driver: 550.120

### Additional context

This bug was detected during live visualization of rewards in the IsaacSim GUI, where a curriculum reward setting was used.  
The `ManagerLiveVisualizer` relies on `_step_reward` to report reward terms visually, and in this case, it incorrectly reported stale reward values when the weight was dynamically changed back to zero.  
This issue affects any tool that uses `_step_reward` for visualization or logging.  

Note that the total reward (`_reward_buf`) computation remains correct, since skipping a term with zero weight or explicitly adding zero yields the same result.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [x] `self._step_reward` is explicitly set to zero for zero-weight terms
- [x] Per-term step reward accurately reflects zero contribution when weight is zero

RandomOakForest, 2025-04-28:
Thank you for posting this. Did you mean Isaac Lab 2.1.0 and Isaac Sim 4.5? Thank you for adding the PR.

bikcrum, 2025-04-28:
> Thank you for posting this. Did you mean Isaac Lab 2.1.0 and Isaac Sim 4.5? Thank you for adding the PR.

That is correct. Thanks for pointing that out. I have updated them.

PR #2392 (2025-05-09): Resets step reward buffer properly when weight is zero
# Description

This pull request fixes a bug where `_step_reward` could retain stale values when a reward term's weight was dynamically changed back to zero.  
Previously, when a reward term had zero weight, the computation skipped updating `_step_reward`, assuming that it would stay correct.  
However, if the weight was first changed from zero to nonzero and then back to zero during runtime (e.g., in curriculum settings), stale nonzero values could persist, causing incorrect reward visualizations or logging.

This change explicitly sets ` [... cut]
commit f1ba9c3a (2025-05-09): Resets step reward buffer properly when weight is zero (#2392)

# Description

This pull request fixes a bug where `_step_reward` could retain stale
values when a reward term's weight was dynamically changed back to zero.
Previously, when a reward term had zero weight, the computation skipped
updating `_step_reward`, assuming that it would stay correct.
However, if the weight was first changed from zero to nonzero and then
back to zero during runtime (e.g., in curriculum settings), stale
nonzero values could persist, causing incorrect reward visualizations or
logging.

This change explicitly s [... cut]
closed 2025-05-09 by Mayankm96