isaac-sim/IsaacLab #2399: [Bug Report] AttributeError: 'NoneType' object has no attribute 'get_model

songtm, 2025-04-29:
### Describe the bug

run the create_empty.py will get the "AttributeError: 'NoneType' object has no attribute 'get_model'" error output

### Steps to reproduce

```
[6.438s] [ext: isaacsim.core.prims-0.3.7] startup
[6.495s] [ext: omni.warp-1.5.0] startup
2025-04-29 08:19:34 [6,447ms] [Error] [carb.scripting-python.plugin] AttributeError: 'NoneType' object has no attribute 'get_model'

At:
  D:\Tools\Omniverse\Library\IsaacLab\_isaac_sim\extscache\omni.kit.browser.sample-1.4.8\omni\kit\browser\sample\extension.py(114): register_sample_folder
  d:/tools/omniverse/library/isaaclab/_isaac_sim/extscache/omni.warp-1.5.0/omni/warp/_extension/__init__.py(206): on_startup
  D:\Tools\Omniverse\Library\IsaacLab\_isaac_sim\kit\kernel\py\omni\ext\_impl\_internal.py(164): _startup_ext
  D:\Tools\Omniverse\Library\IsaacLab\_isaac_sim\kit\kernel\py\omni\ext\_impl\_internal.py(224): startup
  D:\Tools\Omniverse\Library\IsaacLab\_isaac_sim\kit\kernel\py\omni\ext\_impl\_internal.py(328): startup_extension
  PythonExtension.cpp::startup()(2): <module>
  D:\Tools\Omniverse\Library\IsaacLab\\_isaac_sim\exts\isaacsim.simulation_app\isaacsim\simulation_app\simulation_app.py(427): _start_app
  D:\Tools\Omniverse\Library\IsaacLab\\_isaac_sim\exts\isaacsim.simulation_app\isaacsim\simulation_app\simulation_app.py(212): __init__
  D:\Tools\Omniverse\Library\IsaacLab\source\isaaclab\isaaclab\app\app_launcher.py(780): _create_app
  D:\Tools\Omniverse\Library\IsaacLab\source\isaaclab\isaaclab\app\app_launcher.py(131): __init__
  D:\Tools\Omniverse\Library\IsaacLab\scripts\tutorials\00_sim\create_empty.py(29): <module>

2025-04-29 08:19:34 [6,447ms] [Error] [omni.ext.plugin] [ext: omni.warp-1.5.0] Failed to startup python extension.
[6.661s] [ext: omni.kit.window.extensions-1.4.25] startup
[6.676s] [ext: omni.sensors.nv.wpm-2.4.0-coreapi] startup
[6.680s] [ext: isaacsim.core.api-4.2.16] startup
[6.698s] [ext: omni.replicator.core-1.11.35] startup
2025-04-29 08:19:34 [6,779ms] [Warning] [omni.replicator.core.scripts.extension] No material configuration file, adding configuration to material settings directly.
[6.855s] [ext: isaacsim.gui.components-1.0.9] startup

```

### System Info

Describe the characteristic of your environment:

- Commit: 2e6946af
- Isaac Sim Version: 4.5.0-rc.36+release.19112.f59b3005.gl
- OS: win11
- GPU: RTX4060 laptop GPU
- CUDA: 12.6
- GPU Driver: 560.94

### Additional context

None.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

RandomOakForest, 2025-04-29:
Thank you for posting this. Which version of Isaac Lab is this? Thanks.

songtm, 2025-04-29:
> Thank you for posting this. Which version of Isaac Lab is this? Thanks.
IsaacLab 2.1.0;  I just find #563's attach log has this error msg too

RandomOakForest, 2025-06-04:
Thank you for following up. The team will review this.

AntoineRichard, 2026-08-27:
Thanks @songtm for filing this with a full traceback, and sorry for the long silence.

Every failing frame sits in Isaac Sim extension startup (`omni.warp-1.5.0/_extension/__init__.py` → `omni.kit.browser.sample/extension.py`); the only Isaac Lab frames are `app_launcher` creating the `SimulationApp`. That code path lives outside Isaac Lab, and the Isaac Sim 4.5 / `omni.warp` 1.5.0 combination is several releases behind — Isaac Lab now pins `isaacsim==6.0.1.0`. Closing on that basis; please reopen with a fresh log if you still hit it on a current release.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard