isaac-sim/IsaacLab #2400: [Proposal] More configurations for convex decomposition in the mesh converter

hu-hy17, 2025-04-29:
### Proposal

Allow users to specify more configurations when converting mesh files (.obj, .stl, etc.) to USD files through `isaaclab.sim.converters.MeshConverter`.

### Motivation

When I use `MeshConverter` with `--collision-approximation convexDecomposition` option to get a collision model, the collision model is significantly larger than the visual one. 

The problem is that the converter uses a very low `voxel resolution` by default (I think it should be 50000), as shown in the picture (the left one is converted from `MeshConverter` and the right one is re-decomposed with `voxel resolution=5000000`).

[image]

I tried to solve this by modifying lines 124- 132 in `isaaclab.sim.converters.MeshConverter`
```
from pxr import PhysxSchema
if cfg.collision_props is not None:
    # -- Collision approximation to mesh
    mesh_collision_api = UsdPhysics.MeshCollisionAPI.Apply(child_mesh_prim)
    mesh_collision_api.GetApproximationAttr().Set(cfg.collision_approximation)

    # Add the following three lines
    if cfg.collision_approximation == 'convexDecomposition':
        cvx_api = PhysxSchema.PhysxConvexDecompositionCollisionAPI.Apply(child_mesh_prim)
        cvx_api.CreateVoxelResolutionAttr(5000000)  # 50000-5000000

    # -- Collider properties such as offset, scale, etc.
    schemas.define_collision_properties(
        prim_path=child_mesh_prim.GetPath(), cfg=cfg.collision_props, stage=stage
    )
```

I hope that in addition to voxel resolution, users can also define more options to get a better collision model.

### Alternatives

### Additional context

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ]

RandomOakForest, 2025-04-29:
Thank you for posting this proposal. The team will review it.

AntoineRichard, 2026-08-27:
Thanks @hu-hy17 for the request — this is now supported.

`MeshConverterCfg.mesh_collision_props` accepts a `PhysxConvexDecompositionPropertiesCfg` (or a list of fragments), exposing `voxel_resolution`, `error_percentage`, `shrink_wrap`, `max_convex_hulls` and more, applied per child mesh prim. See `source/isaaclab/isaaclab/sim/converters/mesh_converter_cfg.py` and `source/isaaclab_physx/isaaclab_physx/sim/schemas/schemas_cfg.py`; the schema work landed in https://github.com/isaac-sim/IsaacLab/pull/2249. Closing as implemented — please reopen if a cooking parameter you need is still missing.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard