isaac-sim/IsaacLab #2409: Import style mismatch between pip-installed and source-installed versions

shirgur, 2025-04-30:
### Describe the bug

The following is a potential “bug”.
For “Advanced installation” - [link](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/isaaclab_pip_installation.html), the package is packed as if we work in the repo directory tree.
Meaning:
- The repo contains several packages under ```source/```
- When following the installation from source - [link](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html), each package can be imported nicely like ```import isaaclab.assets``` for example
- But when following the advanced installation the import style is different, so for the above example we get ```import isaaclab.source.isaaclab.assets```

### Steps to reproduce

Follow the “Advanced installation” - [link](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/isaaclab_pip_installation.html)

### System Info

Describe the characteristic of your environment:

- Commit: main
- Isaac Sim Version: 4.5.0
- OS: Ubuntu 22.04

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Remove ```isaaclab.source``` prefix when installing using pip

RandomOakForest, 2025-05-05:
Thanks for posting this. Could you let us know which version of Isaac Lab are you using, or which commit? Typically, this is fixed with a clean reinstall.

shirgur, 2025-05-06:
**IsaacLab version**: 2.1.0

Simply follow the "Advanced installation" using this [link](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/isaaclab_pip_installation.html)
BTW - there should be quotes like ```'isaaclab[isaacsim,all]==2.1.0'``` in the pip install command :)

Toni-SM, 2025-06-26:
Hi @shirgur 

> For “Advanced installation” - [link](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/isaaclab_pip_installation.html), the package is packed as if we work in the repo directory tree

Since Isaac Lab is a Omniverse Kit-based appliaction, the `isaaclab` Python package from PyPI is not a Python packages per se, but as an application packaged and shipped as Python packages to support the `pip install` way. 

While you may import using the `isaaclab.source...` style (and it will be detected by IntelliSense), it is a side effect of the required folder structure and therefore not a recomendable/applicable way to construct the import statements. 

Currently, we are working on a solution to setup extraAnalysisPath [properly](https://docs.isaacsim.omniverse.nvidia.com/latest/installation/install_python.html#generating-vs-code-settings) for the Python packages, since IntelliSense does not work out of the box by default.

---

See https://github.com/isaac-sim/IsaacLab/issues/2753#issuecomment-3000742206 and https://github.com/isaac-sim/IsaacLab/issues/2753#issuecomment-3005042295 for more discussions.

closed 2025-06-30 by kellyguo11