isaac-sim/IsaacLab #2416: Bug in source/isaaclab/isaaclab/utils/warp/ops.py ray_distance.view(shape[0], shape[1]) causes shape mismatch

Finn-Sueberkrueb, 2025-05-02:
I am trying to use raycast_mesh(..., return_distance=True) with the following input:

```
ray_starts = torch.randn(4, 3)
ray_directions = torch.randn(4, 3)
```

This results in:

```
ray_distance = ray_distance.to(device).view(shape[0], shape[1])
RuntimeError: shape '[4, 3]' is invalid for input of size 4
```

It seems ray_distance is initialized with shape (N,) where N = 4, but the code attempts to reshape it to (4, 3) which is incorrect for scalar distances.

Shouldn’t the correct line be:

`ray_distance = ray_distance.to(device).view(shape[0])`

instead of:

`ray_distance = ray_distance.to(device).view(shape[0], shape[1])`

Finn-Sueberkrueb, 2025-05-02:
Or in the [Docu](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.utils.html#isaaclab.utils.warp.raycast_mesh) it should say: 

> ray_starts – The starting position of the rays. Shape (N, **1,** 3).
> ray_directions – The ray directions for each ray. Shape (N, **1,** 3).

RandomOakForest, 2025-05-05:
Thanks for catching this. We will review.

closed 2026-09-26 by StafaH