isaac-sim/IsaacLab #2420: [Question] Method to set joints directly without simulating motion?

algetops, 2025-05-03:
Context: I was looking to speed up training for a non-RL task by using IsaacLab's parallel simulation and tensor API. I need to be able to find the relative positions of several rigid bodies in my articulation given a set of joint angles, which I figured I could achieve with FrameTransformers. However, simulating motion between two configurations significantly slows this process down, and depending on articulation parameters it can induce jitter that adds significant noise to the ideal result.

My Question: Does IsaacLab expose methods to directly set joint positions in-sim, as well as disable the velocity/torque based simulation? I essentially want to recreate 'ideal' behaviour; the particular problem I'm working on doesn't really suffer from sim2real issues. I did see [this possibly related issue](https://github.com/isaac-sim/IsaacLab/issues/1631) about computing the forward kinematics of an articulation without updating it, but I would like the FrameTransformers to update so it doesn't seem applicable.

RandomOakForest, 2025-05-05:
Thanks for posting this. Have you tried using the [Frame Transformer](https://docs.omniverse.nvidia.com/kit/docs/omni_physics/latest/extensions/runtime/source/omni.physics.tensors/docs/api/python.html#omni.physics.tensors.impl.api.SimulationView.update_articulations_kinematic) with `omni.physics.tensors` to update articulations kinematics by calling [this method](https://docs.omniverse.nvidia.com/kit/docs/omni_physics/latest/extensions/runtime/source/omni.physics.tensors/docs/api/python.html#omni.physics.tensors.impl.api.SimulationView.update_articulations_kinematic)?

RandomOakForest, 2025-06-04:
Thank you for following up. Please try the recommendation above. If you need further help, please post a new question with details on your case. Thanks.

closed 2025-06-04 by RandomOakForest