isaac-sim/IsaacLab #2425: [Question] Actuator PD tuning issue? possible bug?

Jdowdy05, 2025-05-05:
### Question

I am using Isaac Lab 2.0.2 and Isaac Sim 4.5.0

I am PD tuning the Dynamixel XM430-W350 actuators in Isaac Sim/Lab for the Robotis OP3. Using real-world values for the actuator in any of the actuator configs doesn't give usable results. One example is that increasing my P gain (stiffness) doesn't change my rise time and drastically increases my steady-state error, which I show plots of below. This was using the ideal PD actuator and uses values taken from the spec sheet of the actuator, though I've also tried changing these, and I still cannot achieve the usable results. This doesn't happen with the implicit actuator; however, this just instantaneously achieves the goal position. I can technically achieve real-world performance, but only when torque and velocity limits are 1e+20, but using this for RL leads my small humanoid learning to fly instead of walk. Is this an issue with Isaac sim/lab actuators, or something wrong on my end? 

A few other things I have tried are:
1. Using the single revolute USD and assigning the appropriate mass, inertia, and CoM to the joint 
2. An actuator net from questionable data collection for a current-to-torque LR
3. Modifying the URDF 
4. Trying the MJCF for this robot
5. Switching to IsaacLab 2.1.0

stiffness at 100
[image]

stiffness at 20
([image])

stiffness at 1
[image]

KyleM73, 2025-05-06:
Can you post your robot config? Are you spawning from a urdf or usd? And how have you configured the joint drives, articulation props, etc.? I have also seen some weird behavior from the newer urdf importer but can try to advise if you share the config.

Jdowdy05, 2025-05-06:
@KyleM73 I spawn from a USD, but it's converted from a URDF.  My config can be seen below, along with the rest of my code, which I provide in case I'm performing something incorrectly, but I configure the joint, articulation, and rigid body properties through the standard approach for other robots, as seen in the shown assets. I also do a sanity check and set the values using the provided isaaclab functions, directly setting them in physx at the first environment step. The plot data uses the "head_pan" joint for the real world and the simulation. I did end up getting the implicit actuator to work "good enough".  For whatever reason, if I set the render interval to 1, making it equal to the number of sim physics steps, it no longer has these instant jumps to the goal position (this makes me think the control loop is tied to framerate).  I would still like to get the actuator models in Isaac lab working as they work well for me on other robots. 

Thanks for the reply!

`# Copyright (c) 2022-2025, The Isaac Lab Project Developers.
# All rights reserved.
#
# SPDX-License-Identifier: BSD-3-Clause

""" Used to for PD tuning of articulations.

.. code-block:: bash

    # Usage
    ./isaaclab.sh -p scripts/tutorials/01_assets/run_articulation.py

"""

"""Launch Isaac Sim Simulator first."""

import argparse
from typing import TYPE_CHECKING
import math
from isaaclab.app import AppLauncher

# add argparse arguments
parser = argparse.ArgumentParser(description="Tutorial on spawning and interacting with an articulation.")
# append AppLauncher cli args
AppLauncher.add_app_launcher_args(parser)
# parse the arguments
args_cli = parser.parse_args()

# launch omniverse app
app_launcher = AppLauncher(args_cli)
simulation_app = app_launcher.app

"""Rest everything follows."""

import torch
import numpy as np

import isaacsim.core.utils.prims as prim_utils
from typing import TYPE_CHECKING, Literal

import carb
import isaacsim.core.utils.stage as stage_utils
import omni.log
import omni.physics.tensors.impl.api as physx
from pxr import PhysxSchema, UsdPhysics

import isaaclab.sim as sim_utils
import isaaclab.utils.math as math_utils
import isaaclab.utils.string as string_utils
from isaaclab.actuators import ActuatorBase, ActuatorBaseCfg, ImplicitActuator, ImplicitActuatorCfg, IdealPDActuator, IdealPDActuatorCfg, DCMotor, DCMotorCfg, ActuatorNetMLPCfg
from isaaclab.assets import Articulation, ArticulationCfg, RigidObject, RigidObjectCfg
from isaaclab.managers import SceneEntityCfg
from isaaclab.sim import SimulationContext
from isaaclab.utils.assets import ISAAC_NUCLEUS_DIR

from isaaclab.assets.asset_base import AssetBase
from isaaclab.assets.articulation.articulation_data import ArticulationData

import isaacsim.core.utils.prims as prim_utils

import isaaclab.sim as sim_utils
from isaaclab.assets import Articulation

if TYPE_CHECKING:
    from isaaclab.assets.articulation.articulation_cfg import ArticulationCfg

[... 414 lines cut ...]
    plt.show()

def main():
    """Main function."""
    # Load kit helper
    sim_cfg = sim_utils.SimulationCfg(device=args_cli.device, dt=0.005, render_interval=1)
    sim = SimulationContext(sim_cfg)
    # Set main camera
    sim.set_camera_view([2.5, 0.0, 4.0], [0.0, 0.0, 2.0])
    # Design scene
    scene_entities, scene_origins = design_scene()
    scene_origins = torch.tensor(scene_origins, device=sim.device)
    # Play the simulator
    sim.reset()
    # Now we are ready!
    print("[INFO]: Setup complete...")
    # Run the simulator
    run_simulator(sim, scene_entities, scene_origins)

if __name__ == "__main__":
    # run the main function
    main()
    # close sim app
    simulation_app.close()
`

Jdowdy05, 2025-05-06:
@KyleM73 I spawn from a USD, but it's converted from a URDF.  My config can be seen below, along with the rest of my code, which I provide in case I'm performing something incorrectly, but I configure the joint, articulation, and rigid body properties through the standard approach for other robots, as seen in the shown assets. I also do a sanity check and set the values using the provided isaaclab functions, directly setting them in physx at the first environment step. The plot data uses the "head_pan" joint for the real world and the simulation. I did end up getting the implicit actuator to work "good enough".  For whatever reason, if I set the render interval to 1, making it equal to the number of sim physics steps, it no longer has these instant jumps to the goal position (this makes me think the control loop is tied to framerate).  I would still like to get the actuator models in Isaac lab working as they work well for me on other robots. 

Thanks for the reply!

```python
# Copyright (c) 2022-2025, The Isaac Lab Project Developers.
# All rights reserved.
#
# SPDX-License-Identifier: BSD-3-Clause

""" Used to for PD tuning of articulations.

.. code-block:: bash

    # Usage
    ./isaaclab.sh -p scripts/tutorials/01_assets/run_articulation.py

"""

"""Launch Isaac Sim Simulator first."""

import argparse
from typing import TYPE_CHECKING
import math
from isaaclab.app import AppLauncher

# add argparse arguments
parser = argparse.ArgumentParser(description="Tutorial on spawning and interacting with an articulation.")
# append AppLauncher cli args
AppLauncher.add_app_launcher_args(parser)
# parse the arguments
args_cli = parser.parse_args()

# launch omniverse app
app_launcher = AppLauncher(args_cli)
simulation_app = app_launcher.app

"""Rest everything follows."""

import torch
import numpy as np

import isaacsim.core.utils.prims as prim_utils
from typing import TYPE_CHECKING, Literal

import carb
import isaacsim.core.utils.stage as stage_utils
import omni.log
import omni.physics.tensors.impl.api as physx
from pxr import PhysxSchema, UsdPhysics

import isaaclab.sim as sim_utils
import isaaclab.utils.math as math_utils
import isaaclab.utils.string as string_utils
from isaaclab.actuators import ActuatorBase, ActuatorBaseCfg, ImplicitActuator, ImplicitActuatorCfg, IdealPDActuator, IdealPDActuatorCfg, DCMotor, DCMotorCfg, ActuatorNetMLPCfg
from isaaclab.assets import Articulation, ArticulationCfg, RigidObject, RigidObjectCfg
from isaaclab.managers import SceneEntityCfg
from isaaclab.sim import SimulationContext
from isaaclab.utils.assets import ISAAC_NUCLEUS_DIR

from isaaclab.assets.asset_base import AssetBase
from isaaclab.assets.articulation.articulation_data import ArticulationData

import isaacsim.core.utils.prims as prim_utils

import isaaclab.sim as sim_utils
from isaaclab.assets import Articulation

if TYPE_CHECKING:
    from isaaclab.assets.articulation.articulation_cfg import ArticulationCfg
[... 415 lines cut ...]
    plt.show()

def main():
    """Main function."""
    # Load kit helper
    sim_cfg = sim_utils.SimulationCfg(device=args_cli.device, dt=0.005, render_interval=1)
    sim = SimulationContext(sim_cfg)
    # Set main camera
    sim.set_camera_view([2.5, 0.0, 4.0], [0.0, 0.0, 2.0])
    # Design scene
    scene_entities, scene_origins = design_scene()
    scene_origins = torch.tensor(scene_origins, device=sim.device)
    # Play the simulator
    sim.reset()
    # Now we are ready!
    print("[INFO]: Setup complete...")
    # Run the simulator
    run_simulator(sim, scene_entities, scene_origins)

if __name__ == "__main__":
    # run the main function
    main()
    # close sim app
    simulation_app.close()
```

vicltz, 2025-05-21:
Use _IdealPdActuator_ to ensure consistency between simulators. If your gains are the one of the real system, make sure you added the proper `friction` and `armature` to the joints. It should then be stable.

Jdowdy05, 2025-05-22:
@vicltz Well, that's the issue...changing friction and armature parameters to estimated values from data collection doesn't give accurate results for any actuator model in Isaac Lab. My experience has been that I can never achieve my goal position at any PD, friction, and armature values with low velocity and effort limits.