isaac-sim/IsaacLab #2426: [Question]  PyTorch-compatible forward kinematics outputs?

umang-sarcomere, 2025-05-05:
I have a problem where I'm using the output of a neural network as a set of joint parameters, and the loss function I am using to train it depends on the robot link positions. As such, I need to be able to gradients through the forward kinematics function. So far, I've been using a separate differentiable forward kinematics library to do this so far, but it is seemingly much slower in training than previous projects I have done with Isaac Lab; ideally, I would like to transition my codebase to Lab.

So my question is: is there a way to get differentiable forward kinematics in Isaac Sim? If so, how would I achieve this? I know that the frame jacobians for an Articulation can be retrieved through `Articulation.root_physx_view.get_jacobians()`, but it doesn't seem to play well with a backwards pass in torch.

RandomOakForest, 2025-05-06:
Thank you for posting this. I'll inquire with the team and post back here. Good question.

umang-sarcomere, 2025-05-06:
Thank you! That would be greatly appreciated.

RandomOakForest, 2025-05-14:
Following up on this. Unfortunately, I have confirmed this is not currently possible in Isaac Lab 2.1.0 or earlier. The team is aware of this, and we are constantly working to fill in this features. Thank you for your interest in Isaac Lab.

closed 2025-05-14 by RandomOakForest

mmacklin, 2025-05-14:
Hi there, we do have differentiable forward kinematics available in Warp, this can be connected to Torch through a custom autograd.op for optimization as in this inverse kinematics example:

https://github.com/NVIDIA/warp/blob/main/warp/examples/optim/example_inverse_kinematics_torch.py

(see `eval_fk()` method).

Warp is available in Isaac Lab, although it is not integrated with the Articulation class directly.

umang-sarcomere, 2025-05-20:
@RandomOakForest thank you for the update! I'll keep an eye on future Isaac Lab versions.

@mmacklin The methodology in the Warp example looks like it would work well for my use, thank you for pointing it out! I am not super familiar with using Warp directly, but I'll give it a shot and see how it goes.