isaac-sim/IsaacLab #2427: [Bug Report] Cannot import .stp file into isaacsim

kainism, 2025-05-06:
### Isaac Sim Version
4.5.0 (installed with pip)

### Operating System
Ubuntu 22.04

### GPU Information
Model: NVIDIA GeForce RTX 4090 D
Driver Version: 550.107.02
Topic Description
Detailed Description
Cannot import .stp file in isaacsim.

### Steps to Reproduce
$BASH: conda activate env_isaaclab (or your virtual environment)
$BASH: isaacsim
click "file → import " in Isaacsim GUI and select a .stp file to import.
### Error Messages
2025-04-30 13:15:17 [22,317ms] [Error] [carb] loadPlugin: Library path doesn’t exist: /home/kevin/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/omni/plugins/libcarb.crashreporter-breakpad.plugin.so
2025-04-30 13:15:17 [22,317ms] [Error] [omni.connect.core] Failed to load Carbonite’s carb.crashreporter-breakpad.plugin plugin: -1
2025-04-30 13:15:17 [22,360ms] [Warning] [omni.kit.tool.asset_importer.importers_manager] Failed to convert assets [‘/home/kevin/Downloads/CX001_model/rb001_asm_simplified.stp’] since importer failed with [Errno 2] No such file or directory: ‘/home/kevin/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/omni/kit’.

### Tried Moves
Find plugins are in /home/kevin/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/omni/kernel/plugins/ with $BASH: find / -name “libcarb.crashreporter-breakpad.plugin.so”.
Create a symbolic link with bellow script,
$BASH: ln -s /home/kevin/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/omni/kernel/plugins/ /home/kevin/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/omni/plugins/

### Error Messages
2025-04-30 13:18:38 [258,891ms] [Error] [omni.connect.core] Invalid host name - [omniverse://]
2025-04-30 13:18:38 [258,892ms] [Warning] [omni.kit.material.library.material_library] /exts/omni.kit.material.library/lib_paths has changed to “[‘/home/kevin/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/omni/mdl/core/Base’, ‘/home/kevin/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/omni/mdl/core/Volume’]” reloading subid & rebuilding menus
2025-04-30 13:18:38 [259,037ms] [Warning] [omni.kit.tool.asset_importer.importers_manager] Failed to convert assets [‘/home/kevin/Downloads/CX001_model/rb001_asm_simplified.stp’] since importer failed with [Errno 2] No such file or directory: ‘/home/kevin/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/omni/kit’.

RandomOakForest, 2025-05-06:
Thanks for posting this. The team will review this.

celestialdr4g0n, 2025-05-07:
I recommend you use Onshape to import and rigging your robot. The robot does not work out of the box by just import stp file. Or if you had the urdf file of your robot, please use it.

kainism, 2025-05-08:
> The robot does not work out of the box by just importing stp file. Or if you had the urdf file of your robot, please use it.

Thanks for your reply! I do have a .urdf file which converted from .stp file by solidworks, so my original workflow is ".stp -> .urdf -> .usd". I have 2 questions, 
1. Can I just convert .stp file to a  .usd file which can be directly used in isaaclab?
2. Using the URDF IMPORTER in isaacsim, seems not to have the option "Create Instanceable Asset" mentioned in https://docs.isaacsim.omniverse.nvidia.com/4.5.0/isaac_lab_tutorials/tutorial_instanceable_assets.html, as shown below.

[image]

celestialdr4g0n, 2025-05-08:
1. Yes, you can convert a std to usd file. But it is not enough, for a robot to work it need to have joints connected between the links. And that information cannot pass between stp and usd. The benefit of using urdf is Isaac Sim can use the information in urdf and construct the joints between the links which has their mesh stored in stp or dea format.
I recommend you learn IsaacSim and suffer like me before touching IsaacLab. Because IsaacLab is built on top of Isaac Sim.
2. Instanceable or not depend on your setting in Isaac Sim/ USD. First check your prim structure then tick the Instanceable box.

[image]

closed 2025-07-08 by kellyguo11