isaac-sim/IsaacLab #2430: [Question] Get historical sensor data

torydebra, 2025-05-06:
### Question

I see the [contact sensor](https://isaac-sim.github.io/IsaacLab/main/source/overview/core-concepts/sensors/contact_sensor.html) has a `history_length` for getting past data, but how is this accessed? 

When getting the data with `scene["contact_forces_H"].data.force_matrix_w`, only the current data (a single 3D vector) is provided for each environment, for each contact sensor.

RandomOakForest, 2025-05-06:
Thanks for posting this. For the contact sensor, it should be `._data`, which is an instance of `ContactSensorData` ([ref.](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.sensors.html#isaaclab.sensors.ContactSensorData)).

torydebra, 2025-05-06:
Thanks for the answer, 
indeed I am using data (without underscore right?), but I see only the current occurency of the data:

eg adding 
```
print("contact forces full")
print(scene["contact_forces"].data.net_forces_w)
```
in https://github.com/isaac-sim/IsaacLab/blob/9be0de5f6196374fa793dc33978434542e7a40d3/scripts/tutorials/04_sensors/add_sensors_on_robot.py#L154

and running with `./isaaclab.sh -p scripts/tutorials/04_sensors/add_sensors_on_robot.py --num_envs 2 --enable_cameras`

it prints
```
contact forces full
tensor([[[1.6603e-05, 0.0000e+00, 1.2357e+02],
         [1.7969e-05, 0.0000e+00, 1.3374e+02],
         [1.7514e-05, 0.0000e+00, 1.3035e+02],
         [1.6618e-05, 0.0000e+00, 1.2368e+02]],

        [[1.6765e-05, 0.0000e+00, 1.2478e+02],
         [1.8409e-05, 0.0000e+00, 1.3702e+02],
         [1.7421e-05, 0.0000e+00, 1.2966e+02],
         [1.6355e-05, 0.0000e+00, 1.2173e+02]]], device='cuda:0')
```

which is, 2 environment, 4 sensors (one for each leg), despite `history_length=6`

ozhanozen, 2025-05-06:
You need to use `net_forces_w_history` rather than `net_forces_w`.

https://github.com/isaac-sim/IsaacLab/blob/9be0de5f6196374fa793dc33978434542e7a40d3/source/isaaclab/isaaclab/sensors/contact_sensor/contact_sensor_data.py#L45-L57

However, this does not allow you to distinguish the forces according to the filtered bodies, as `force_matrix_w` would. For this have a look at #1746.

torydebra, 2025-05-06:
thanks, I see now.

Related question: is the air_time/contact_time considering the filtered bodies or not?

ozhanozen, 2025-05-06:
I would say filtered bodies are not considered separately based on this (so any contact affects):

https://github.com/isaac-sim/IsaacLab/blob/9be0de5f6196374fa793dc33978434542e7a40d3/source/isaaclab/isaaclab/sensors/contact_sensor/contact_sensor.py#L357

closed 2025-05-08 by torydebra