isaac-sim/IsaacLab #2432: [Bug Report] Type checking for callable classes in MDP terms

KyleM73, 2025-05-06:
Static type checkers complain about callable classes when being used for MDP terms. Specifically:

`func: Callable[..., torch.Tensor] = myClass`

fails the type check because myClass does not return the desired type. Even if the desired type is returned by the `__call__` method, the type checker would only pass if it was instantiated as 

`func: Callable[..., torch.Tensor] = myClass()`

but this is obviously undesirable because we do not want to have to instantiate every class when constructing the config file.

Is there a workaround? I would prefer not to litter my code with `# type: ignore` everywhere. This isn't a huge deal obviously, the code still works correctly, it's just a minor annoyance. Thanks!

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

- [ ] callable MDP terms play nice with static type checkers

RandomOakForest, 2025-05-07:
Thanks for posting this. The team will review this. In the meantime, maybe using an explicit return type annotation works?
```
class MyClass:
    def __call__(self, *args) -> torch.Tensor:  # Explicit return type
        ... 
```
Alternatively, you may be able to leverage _Protocols_ for callable classes.

Mikasa0503, 2026-09-27:
On current develop, the typed func fields in ObservationTermCfg and other MDP term configs still exclude ManagerTermBase subclasses, while ManagerBase explicitly accepts and instantiates them. A small standalone Pyright probe reproduces the class-versus-callable return-type mismatch; it is not an Isaac Lab runtime test.

Would a focused typing-only fix using Callable[...] | type[ManagerTermBase], as suggested earlier in this issue, still be welcome? I would keep runtime behavior unchanged and verify the supported function/class cases against the actual package. Since this issue is assigned and configuration cleanup is active, please let me know if this is already being handled or should be included in that work instead.

closed 2026-10-02 by AntoineRichard