isaac-sim/IsaacLab #2433: [Question] Request for the GR1T2_fourier_hand_6dof.usd asset.

gaolin131, 2025-05-07:
The online assets are very laggy. This asset is not included in the asset package. Can you send it out separately?

RandomOakForest, 2025-05-07:
Thanks for posting this. Are you using [asset caching](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/asset_caching.html)? When do you observe the lag?

gaolin131, 2025-05-07:
> Thanks for posting this. Are you using [asset caching](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/asset_caching.html)? When do you observe the lag?感谢您发送编修。您是否使用[资产缓存](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/asset_caching.html)？你什么时候观察到滞后？

I have [downloaded the asset](https://docs.isaacsim.omniverse.nvidia.com/latest/installation/download.html) to my local machine; however, the downloaded asset does not include GR1T2_Fourier_hand_6df.USD. Additionally, loading the official online asset results in significant lag. Could you please provide the compressed package for this asset?

qingcaicai777, 2025-05-26:
Pls is the problem solved, I am also having the same problem and at the same time I am also having the problem that the file cannot be found packing_table.usd

AntoineRichard, 2026-08-27:
Thanks for flagging this @gaolin131 — you were right that the file was missing.

`GR1T2_fourier_hand_6dof.usd` genuinely wasn't in the Isaac Sim 4.5 asset pack you were using, but it ships in the current packs: `Robots/FourierIntelligence/GR-1/GR1T2_fourier_hand_6dof/GR1T2_fourier_hand_6dof.usd` resolves fine under `Assets/Isaac/5.0` and `Assets/Isaac/6.1` (the same goes for `Props/PackingTable/packing_table.usd`). Re-downloading or re-pointing your asset root at the pack matching your Isaac Sim version should sort it — see the [installation docs](https://isaac-sim.github.io/IsaacLab/release/3.0.0/source/setup/installation/index.html) for asset caching. Reopen if it's still 404-ing.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard