isaac-sim/IsaacLab #2434: How to modify restitution coefficient of RigidBodyObject in IsaacLab?

explore-code-world7, 2025-05-07:
### Question

Basic questions, related to robot learning, that are not bugs or feature requests will be closed without reply, because GitHub issues are not an appropriate venue for these.

Advanced/nontrivial questions, especially in areas where documentation is lacking, are very much welcome.

For questions that are related to running and understanding Isaac Sim, please post them at the official [Isaac Sim forums](https://docs.omniverse.nvidia.com/app_isaacsim/app_isaacsim/isaac_sim_forums.html).

### Isaac Sim Version
[x] 4.5.0

### Operating System
[x] Ubuntu 22.04

### GPU Information
* Model: NVIDIA GeForce RTX 4060
* Driver Version: 550.144.03

## Topic Description
### Detailed Description
I can access and modify restitution coefficient for both terrains and rigid_objects in IsaacGym.
I can do the same things to terrains in IsaacLab too, 
but I tried to check many class configuring files in the source category of IsaacLab repository,
still don't know how to access and modify restitution coefficient of rigid object in IsaacLab.
Can anyone give a help, thank you a lot!

In IsaacGym, 
1. as the example code file(IsaacGym/python/examples/body_physics_props.py) shows,
the restitution attribute of rigid object can be easily accessed via the returned class  of function get_actor_rigid_shape_properties() —— [isaacgym.gymapi.RigidShapeProperties](https://junxnone.github.io/isaacgymdocs/api/python/struct_py.html).
[image]

```python
        # add box actor
        pose = gymapi.Transform()
        pose.p = gymapi.Vec3(0.0, 0.5, 0.0)
        pose.r = gymapi.Quat(0, 2, 2, 1)        # 怎么改变四元数呢?
        box_handle = gym.create_actor(env, asset_box, pose, "actor1", i, 0) # actor 创建的时候要制定位置和姿态
        actor_handles.append(box_handle)

        # set restitution for box actor
        shape_props = gym.get_actor_rigid_shape_properties(env, box_handle)
        shape_props[0].restitution = 1.0
        shape_props[0].compliance = 0.5
        gym.set_actor_rigid_shape_properties(env, box_handle, shape_props)

        # add capsule actor
        pose.p = gymapi.Vec3(0.0, 2.0, 0.0)
        capsule_handle = gym.create_actor(env, asset_capsule, pose, "actor2", i, 0)
```
2. the restitution exponent of a terrain can also be modified with class [isaacgym.gymapi.PlaneParams](https://junxnone.github.io/isaacgymdocs/api/python/struct_py.html)
[image]
[image]
[image]

In IsaacLab, similarly, the restitution attribute of terrain, which is often configured by RigidBodyMaterialCfg can be accessed.

[members of isaaclab.sim.spawners.materials.RigidBodyMaterialCfg](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.assets.html#isaaclab.sim.spawners.materials.RigidBodyMaterialCfg)

[image]

An example is as follows:
```python
    terrain = TerrainImporterCfg(
        prim_path="/World/ground",
        terrain_type="plane",
        collision_group=-1,
        physics_material=sim_utils.RigidBodyMaterialCfg(
            friction_combine_mode="average",
            restitution_combine_mode="average",
            # friction coefficient
            static_friction=1.0, 
            dynamic_friction=1.0,
            # elastic coefficient
            restitution=0.0,
        ),
        debug_vis=False,
    )
```

However, I didn't found that how to access the restitution attribute of a rigid_body or articulation.

[members of isaaclab.assets.RigidObject](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.assets.html#isaaclab.assets.RigidObject.cfg)

[members of isaaclab.assets.RigidObjectCfg](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.assets.html#isaaclab.assets.RigidObjectCfg)

[members of isaaclab.sim.spawners.SpawnerCfg](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.assets.html#isaaclab.sim.spawners.SpawnerCfg)

[members of isaaclab.sim.spawners.RigidObjectSpawnerCfg](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.assets.html#isaaclab.sim.spawners.SpawnerCfg.func)

[members of isaaclab.sim.spawners.from_files.UsdFileCfg](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.assets.html#isaaclab.sim.spawners.from_files.UsdFileCfg)

An example([IsaacLab/scripts/tutorials/01_assets/run_rigid_object.py](https://github.com/isaac-sim/IsaacLab/blob/main/scripts/tutorials/01_assets/run_rigid_object.py)) is as follows:

```python
    cone_cfg = RigidObjectCfg(
        prim_path="/World/Origin.*/Cone",
        spawn=sim_utils.ConeCfg(
            radius=0.1,
            height=0.2,
            rigid_props=sim_utils.RigidBodyPropertiesCfg(),
            mass_props=sim_utils.MassPropertiesCfg(mass=1.0),
            collision_props=sim_utils.CollisionPropertiesCfg(),
            visual_material=sim_utils.PreviewSurfaceCfg(diffuse_color=(0.0, 1.0, 0.0), metallic=0.2),
        ),
        init_state=RigidObjectCfg.InitialStateCfg(),
    )
    cone_object = RigidObject(cfg=cone_cfg)
```

I can't figure out how to modify the restitution coefficient of rigid object~

RandomOakForest, 2025-05-13:
Thank you for posting this. Unfortunately, you will need to create a PhysX material (having `physics_material`) and then bind it to the object, so that you can then use it with `RigidObject`.

explore-code-world7, 2025-05-14:
> Thank you for posting this. Unfortunately, you will need to create a PhysX material (having `physics_material`) and then bind it to the object, so that you can then use it with `RigidObject`.

Thank your timely reply! I'm sorry to inform your that the formal problem is caused by my negligence. But I have another problem.

## Solution for modify restitution coefficient of rigid object
I found that I can modify restitution coefficient of rigid object with just [ShapeCfg](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.sim.spawners.html#module-isaaclab.sim.spawners.shapes), which inherits [RigidObjectSpawnerCfg](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.sim.spawners.html#spawners.
[image]

Since ShapeCfg and all its subclasses have member variable physicals_meterial, the corresponding restitution can be configured.
[image]

## Problem——How to modify restitution coefficient of [UsdFileCfg](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.sim.spawners.html#isaaclab.sim.spawners.from_files.UsdFileCfg)

[image]

By preliminary inference, I'm likely to check rigid_props or collision_props for restiutition access.