isaac-sim/IsaacLab #2437: [Question] Deploying trained humanoid policy in Isaac Sim

MrOCW, 2025-05-07:
### Question

I have trained a G1 locomotion policy (1), with 13 action output by removing the ActuatorCfg in G1_CFG instead of the default 37 and modified the joints in ActionsCfg of velocity_env_cfg.py. This is to prevent the policy from controlling the arms as I need to control the arms for other purposes.
The training in Isaac Lab went well. Note that the observation size is still 37.

When attempting to deploy policy 1, i used the H1FlatTerrainPolicy in the Humanoid Examples as a template. Damping and Stiffness of all the joints were adjusted according to the env.yaml file. I printed self.robot.dof_names

1. in forward method, I modified self.action to be of length 37 before passing it into ArticulationAction
Sample self.robot.dof_names & self.action:
```
['torso_joint', 'left_shoulder_pitch_joint', 'right_shoulder_pitch_joint', 'left_hip_pitch_joint', 'right_hip_pitch_joint', 'left_shoulder_roll_joint', 'right_shoulder_roll_joint', 'left_hip_roll_joint', 'right_hip_roll_joint', 'left_shoulder_yaw_joint', 'right_shoulder_yaw_joint', 'left_hip_yaw_joint', 'right_hip_yaw_joint', 'left_elbow_pitch_joint', 'right_elbow_pitch_joint', 'left_knee_joint', 'right_knee_joint', 'left_elbow_roll_joint', 'right_elbow_roll_joint', 'left_ankle_pitch_joint', 'right_ankle_pitch_joint', 'left_ankle_roll_joint', 'right_ankle_roll_joint', 'left_five_joint', 'left_three_joint', 'left_zero_joint', 'right_five_joint', 'right_three_joint', 'right_zero_joint', 'left_six_joint', 'left_four_joint', 'left_one_joint', 'right_six_joint', 'right_four_joint', 'right_one_joint', 'left_two_joint', 'right_two_joint']
[ 0.01355542  0.          0.          0.23275226 -0.2605567   0.
  0.          0.17375039  0.07851431  0.          0.         -0.0086574
 -0.18230547  0.          0.          0.48574701  0.82196361  0.
  0.          0.94566303  3.32378697 -0.11683509  1.14421988  0.
  0.          0.          0.          0.          0.          0.
  0.          0.          0.          0.          0.          0.
  0.        ]
```

I get this error message:
```
2025-05-07 09:37:11 [8,712,897ms] [Error] [omni.kit.app._impl] [py stderr]: Traceback (most recent call last):
2025-05-07 09:37:11 [8,712,897ms] [Error] [omni.kit.app._impl] [py stderr]:   File "c:/isaacsim/exts/isaacsim.examples.interactive/isaacsim/examples/interactive/humanoid/humanoid_example.py", line 255, in on_physics_step
2025-05-07 09:37:11 [8,712,898ms] [Error] [omni.kit.app._impl] [py stderr]:     self.h1.initialize()
2025-05-07 09:37:11 [8,712,898ms] [Error] [omni.kit.app._impl] [py stderr]:   File "c:/isaacsim/exts/isaacsim.robot.policy.examples/isaacsim/robot/policy/examples/robots/g1.py", line 387, in initialize
2025-05-07 09:37:11 [8,712,898ms] [Error] [omni.kit.app._impl] [py stderr]:     return super().initialize(set_articulation_props=False)
2025-05-07 09:37:11 [8,712,898ms] [Error] [omni.kit.app._impl] [py stderr]:   File "c:/isaacsim/exts/isaacsim.robot.policy.examples/isaacsim/robot/policy/examples/controllers/policy_controller.py", line 109, in initialize
2025-05-07 09:37:11 [8,712,899ms] [Error] [omni.kit.app._impl] [py stderr]:     self.robot._articulation_view.set_gains(stiffness, damping)
2025-05-07 09:37:11 [8,712,899ms] [Error] [omni.kit.app._impl] [py stderr]:   File "c:/isaacsim/exts/isaacsim.core.prims/isaacsim/core/prims/impl/articulation.py", line 3035, in set_gains
2025-05-07 09:37:11 [8,712,900ms] [Error] [omni.kit.app._impl] [py stderr]:     stiffnesses = self._backend_utils.assign(
2025-05-07 09:37:11 [8,712,900ms] [Error] [omni.kit.app._impl] [py stderr]:   File "c:/isaacsim/exts/isaacsim.core.utils/isaacsim/core/utils/numpy/tensor.py", line 91, in assign
2025-05-07 09:37:11 [8,712,900ms] [Error] [omni.kit.app._impl] [py stderr]:     dst[tuple(indices)] = src
2025-05-07 09:37:11 [8,712,900ms] [Error] [omni.kit.app._impl] [py stderr]: ValueError: shape mismatch: value array of shape (13,) could not be broadcast to indexing result of shape (1,37)
```
The G1 falls immediately upon playing it is observed to still respond to commands.

2. Went to torso_link and excluded the arms from articulation. modified self.robot.get_joint_positions() and self.robot.get_joint_velocities() such that they match the order of self.robot_dof_names and padded with 0s to length 37.
Sample self.robot.dof_names & self.action:
```
['left_hip_pitch_joint', 'right_hip_pitch_joint', 'torso_joint', 'left_hip_roll_joint', 'right_hip_roll_joint', 'left_hip_yaw_joint', 'right_hip_yaw_joint', 'left_knee_joint', 'right_knee_joint', 'left_ankle_pitch_joint', 'right_ankle_pitch_joint', 'left_ankle_roll_joint', 'right_ankle_roll_joint']
[-0.11803413 -1.2521374  -0.12940435  0.04658721  0.52817    -0.24584377
 -0.43887097 -0.21444759  0.9849761   2.5686626  -1.6447524  -3.0439992
  0.9085994 ]
```
I assume this should be the order used in IsaacLab? Otherwise, where is the right place to check?

The error from (1) is gone but the outcome is the same.

I then attempted to train another policy by using the g1_minimal.usd from isaac sim assets, and excluding the arms from articulation.
rsl_rl crashes midway during training regarding a torch.normal error requiring std>=0, and skrl fails to train, with NaN values in the rewards.

env.yaml:
```
sim:
  physics_prim_path: /physicsScene
  device: cuda:0
  dt: 0.005
  render_interval: 4
  gravity: !!python/tuple
  - 0.0
  - 0.0
  - -9.81
  enable_scene_query_support: false
  use_fabric: true
  physx:
    solver_type: 1
    min_position_iteration_count: 1
    max_position_iteration_count: 255
    min_velocity_iteration_count: 0
[... 150 lines cut ...]
        damping:
          .*_hip_yaw_joint: 5.0
          .*_hip_roll_joint: 5.0
          .*_hip_pitch_joint: 5.0
          .*_knee_joint: 5.0
          torso_joint: 5.0
        armature:
          .*_hip_.*: 0.01
          .*_knee_joint: 0.01
          torso_joint: 0.01
        friction: null
      feet:
        class_type: isaaclab.actuators.actuator_pd:ImplicitActuator
        joint_names_expr:
        - .*_ankle_pitch_joint
        - .*_ankle_roll_joint
        effort_limit: 20
        velocity_limit: null
        effort_limit_sim: null
        velocity_limit_sim: null
        stiffness: 20.0
        damping: 2.0
        armature: 0.01
        friction: null
```

RandomOakForest, 2025-05-07:
Thanks for posting this. It just looks like a bad shape in one your input arrays. To be on the same page, could you share a link to the examples you are drawing from? Also, please share your Isaac Lab and Isaac Sim versions. Thanks.

MrOCW, 2025-05-07:
Isaac Lab (main: 9be0de5f6196374fa793dc33978434542e7a40d3)
Isaac Sim 4.5.0
isaacsim/exts/isaacsim.examples.interactive/isaacsim/examples/interactive/humanoid/humanoid_example.py
please also look for H1FlatTerrainPolicy class in another file
I basically created and replaced H1FlatTerrainPolicy with a new G1FlatTerrainPolicy, which loads an rsl_rl jit policy trained in IsaacLab

MrOCW, 2025-05-13:
@RandomOakForest hi! any advise on this problem?